cyoung/stratux #587: AHRS 0.4 beta only works with XINY MPU9250 BMP280 10DOF Acceleration Gyroscope when UBS GPS Vk-162 is unpluged.

ghainbach, 2017-04-21:
1. Stratux version: Raspberry Pi 3 Model B.
2. Stratux config: stratux-ahrs0.4-adfa54813a.img

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: Stratux Vk-162 Remote Mount USB GPS. 

    AHRS
      - [X] yes
      - [ ] no

    power source: EasyAcc 6000mAh Ultra-Slim.

    usb cable: supplied with EasyAcc 6000mAh Ultra-Slim.

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2) iOS 10.3.1

    EFB hardware: (e.g., iPad Mini 2) iPad Mini 2

4. Description of your issue: 

Installed a XINY MPU9250 BMP280 10DOF Acceleration Gyroscope board, and updated to stratux-ahrs0.4-adfa54813a.img (ef8064c09f) for AHRS Beta testing. When GPS/AHRS web page is selected, the AHRS information is blank. When I unplug the USB GPS Vk-162 the AHRS data is correctly displayed in the with the moving paper airplane on the GPS map web page. 

I have also tried "update-stratux-ahrs0.5.1-1fca793b2c.sh" but no AHRS data is displayed with or without the USB GPS Vk-162 installed.  

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-04-21:
Give [0.5.1](http://updates.stratux.me/builds/update-stratux-ahrs0.5.1-1fca793b2c.sh) a try.

ghainbach, 2017-04-21:
[stratux.zip](https://github.com/cyoung/stratux/files/947492/stratux.zip)
stratux.log attached from testing with ahrs0.4-adfa54813a.img.

ghainbach, 2017-04-21:
Hi @cyoung,
Thanks for all the help as always. I installed update-stratux-ahrs0.5.1-1fca793b2c.sh and was still having issues with my iPad Mini 2, and then I connected with a PC running Win7 and IE11 and all was working!  The issue on the iPad was solved by "Settings>Safari>Clear History, Cookies and Data." and then all the AHRS data is now being displayed. Nice work .... I like the G-Meter!

closed 2017-04-23 by westphae

cyoung, 2017-04-24:
@ghainbach - thanks for following up!