cyoung/stratux #588: GPS/AHRS page display with home screen icon

MSGTFE, 2017-04-25:
1. Stratux version: 
stratux-v1.2r2-bcd62e7c00 
ahrs0.5.1-1fca793b2c.sh
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: RY836AI

    AHRS
      - [ ] yes
      - [X] no RY836AI is not connected 

    power source: CANA KIT 5V 2.5 amp

    usb cable: NA

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
ForeFlight 9.0.1
IOS 10.3.1
IPad Mini 2
 
4. Description of your issue:
The new GPS/AHRS page shows up correctly when using Safari and typing in the Stratux address however when using the add to home screen and creating the icon it will not display properly anymore. The rest of the menu pages appear correctly.  Also the attitude and altitude sensor switches are missing as well as the AHRS orientation when trying to access it from the icon.

This only happens when using the home screen icon. When accessing the Stratux UI directly from Safari it displays correctly.

I thought it was a history issue, I have deleted the icon and am unable to get it to display correctly.  

also one other note the MB in  the free storage header is displaying as MiB.

[image]

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-05-01:
Cache manifest needs revisiting.

westphae, 2017-05-10:
Is there any reason to cache anything at all?  By definition the users of this app will be connected to the server, so there cannot be an instance of "offline" use; and the files are all very small.

cyoung, 2017-05-17:
@westphae - load speed, that's about it.

FlyingSikh, 2017-05-26:
1.Stratux version:
 stratux-v1.3b2 (95fe19902a)
update-stratux-v1.3b1-a6f9aaf9f5

2.Stratux config:

SDR
 dual

GPS
 yes
 type: GPYes 

AHRS
 yes

power source: CANA KIT 5V 2.5 amp

usb cable: NA

3.EFB app and version: (e.g., WingX Pro7 8.6.2)
FlyQ most recent

4.Description of your issue:
 The new GPS/AHRS page shows up correctly when using Safari and typing in the Stratux address however when using the add to home screen and creating the icon it will lock up when GPS/AHRS menu is visited the second time. The rest of the menu pages appear correctly. I can visit all other munus from the list and skip GPS/AHRS and it keeps working, soonest I click the GPS/AHRS the 2nd time, it locks up.

This only happens when using the home screen icon. When accessing the Stratux UI directly from Safari it displays correctly.

I thought it was a history issue, I have deleted the icon and am unable to get it to display without freezing

also on my system it also shows in the free storage header is displaying as MiB.

cyoung, 2018-01-17:
@MSGTFE - MiB is right.

I'm not sure what the problem was with the desktop icon and how it related to the browser cache. The best fix right now (AFAIK) is to do exactly as you did -- delete it and type in the address in the browser.

closed 2018-01-17 by cyoung