cyoung/stratux #590: GPS position updates malformed when ownship filtering applied

egid, 2017-04-30:
1. Stratux version: 1.2
2. Stratux config:

    Raspberry Pi 2

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: VK-162, VK-172

    AHRS
      - [ ] yes
      - [X] no

    power source: ship

    usb cable: anker powerline

3. EFB app and version: ForeFlight latest

    EFB platform: iOS 10

    EFB hardware: iPad Mini 4

4. Description of your issue:

I've discussed this with @cyoung already but wanted to get it tracked.

When adding my own (Mode-C only) aircraft's TIS-B position to the ownship filtering in Stratux settings, GPS track and position data become highly erratic (see attached CloudAhoy track). When it's set to a non-target (F00000), performance returns to expected levels.

[image]

cyoung, 2017-05-01:
Looks like FF's position source switching between GPS and received ownship:

https://github.com/cyoung/stratux/blob/master/main/traffic.go#L226

cyoung, 2017-05-24:
Ownship is now filtered as a traffic target but the GDL90 Ownship report is not modified now. Should be fixed in the latest builds.

closed 2017-05-24 by cyoung