cyoung/stratux #594: Adafruit GPS over GPIO support

Austinthemighty, 2017-05-11:
so I have figure out how to get the adafruit GPS to work over the GPIO, the only problem is that the way you set it up on the Pi 1 and 2 differs from the Pi 3

here are the steps that you need to follow for the Pi 1 and 2

1.  Plug in an ethernet cable to pi you will need the internet for this
2. Open up an ssh connection or in terminal run` ifup eth0`
3. Hook the GPS to the Pi as follows
         `VCC to pin 2`
         `GND to pin 6`
         `TX to pin 10`
        ` RX to pin 8`
[image]
4. type in `sudo apt-get install gpsd gpsd-clients python-gps`
5. `sudo systemctl stop gpsd.socket`
    `sudo systemctl disable gpsd.socket`
6.  `sudo nano /boot/cmdline.txt` 
7. deleate everything and type in `dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait`
8. `sudo systemctl stop serial-getty@ttyAMA0.service`
    `sudo systemctl disable serial-getty@ttyAMA0.service `
9. `sudo reboot`
10. `sudo killall gpsd`
      `sudo gpsd /dev/ttyAMA0 -F /var/run/gpsd.sock`
11. Connect to stratux and go to 192.168.10.1 in chrome and check the GPS tab is working

here are the steps that you need to follow for the Pi 3
1.  Plug in an ethernet cable to pi you will need the internet for this
2. Open up an ssh connection or in terminal run` ifup eth0`
3. Hook the GPS to the Pi as follows
         `VCC to pin 2`
         `GND to pin 6`
         `TX to pin 10`
        ` RX to pin 8`
[image]
4. type in `sudo apt-get install gpsd gpsd-clients python-gps`
5. `sudo systemctl stop gpsd.socket`
    `sudo systemctl disable gpsd.socket`
6.  `sudo nano /boot/cmdline.txt` 
7. deleate everything and type in `dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait`
8. `sudo systemctl stop serial-getty@ttyS0.service`
    `sudo systemctl disable serial-getty@ttyS0.service `
9. go to `sudo nano /boot/config.txt` and at the very bottom add `enable_uart=1`
10. `sudo reboot`
11. `sudo killall gpsd`
      `sudo gpsd /dev/ttyS0 -F /var/run/gpsd.sock`
12. Connect to stratux and go to 192.168.10.1 in chrome and check the GPS tab is working

Austinthemighty, 2017-05-17:
Just finished doing a test run, I ran the gps and the pi for around a week and it seems to be fine

cyoung, 2017-05-17:
@Austinthemighty - thanks for this.

I've moved your writeup to the wiki, [here](https://github.com/cyoung/stratux/wiki/Adafruit-Ultimate-GPS-Breakout). If you have further findings or tweaks, go ahead and add them there. Thanks!

closed 2017-05-17 by cyoung