cyoung/stratux #599: Choices for a fan PWM Frequency

lukepalmer, 2017-05-14:
Moving a discussion over from #595 

In the current codebase the fan PWM frequency is 150kHz. I think this is probably too high but am going to follow up with some actual measurements. @cyoung says most people are using an AO3400 MOSFET for switching.

Will post more here later.

cyoung, 2017-05-17:
OK, I got a PI-FAN LD3007MS. The fan acts pretty strange. It seems like it needs to warm up before it even responds to the higher PWM frequency. Sometimes it won't even start at 256/256 with the frequency set at 150 kHz. So I've introduced a failsafe because a number of users that are testing v1.3b2 have reported the same issue.

lukepalmer, 2017-05-17:
If the fan doesn't start at 256/256 then something is wrong- there is no PWM happening at all there, that's just 100% on.

Can you tell me more? Happy to take a look. Ordered some of the FETs most folks are using but that'll take a week or two.

pp4life, 2017-05-19:
I have a 25x25x7mm 5V 0.1A DC Brushless fan. Model FSY27S05L. I just tried update-stratux-v1.3b2-0678955b01.sh which I think Chris said has some code to turn the fan PWN all the way up if the temperature doesn't start coming down. I put the box in front of a heater and the CPU got to 68C. I tried giving the fan a spin by hand, but it did not start. I then shutdown the Stratux, removed the AHRS/Fan control board, and plugged the fan directly into the 5V pins. As soon as I connected power, the fan started up. Does that help any?

cyoung, 2017-05-19:
Did you give it ~10 minutes or so?

lukepalmer, 2017-05-20:
@pp4life I have a few ideas for you. First though, would you mind telling me more about the fan control board you mentioned?

Helno, 2017-05-22:
This is the board being talked about. 
https://github.com/stratux/ahrs/

It will probably be the most common implementation of fan control other than the DIY types like me.

pp4life, 2017-05-23:
@cyoung I did not give it 10 minutes. I can do the test again and wait longer. @Helno thanks for answering @lukepalmer's question about the AHRS/fan control board.

lukepalmer, 2017-05-23:
Hmm shouldn't need 10 minutes. I'll make some measurements using that hardware and come back with a recommendation.

Also, if there is a forum, etc for the AHRS stuff, I'd love to hear about it.

pp4life, 2017-05-23:
https://stratuxadsb.slack.com  There's an AHRS channel on there.

cyoung, 2017-05-23:
@lukepalmer - https://github.com/cyoung/stratux/blob/master/main/fancontrol.go#L74-L81
Works out to about 10 minutes for one cycle. Could take longer.

lukepalmer, 2017-05-23:
Yes. I'm not aware of what effect that code could have though. A PWM duty cycle of 100% is exactly equivalent to setting the output high.

cyoung, 2017-05-23:
Two tests here. In both cases "pwmDuty" was 250 (of 256).

pwmClockDivisor=128:

[image]

pwmClockDivisor=1:

[image]

Doesn't seem to work as we're expecting.

cyoung, 2017-05-23:
Oh, this was measured at "GPIO1" (pin 12).

daveholland86, 2017-05-24:
Just dropping by to say I have this same board AHRS + Fan Control.  Built the Stratux tonight and the fan barely runs.

lukepalmer, 2017-05-24:
I did some work on this and have a fairly satisfying explanation for why some people see bad behavior. Should have a PR in the next day or two along with writeup.

@daveholland86 the fan barely running is a feature, not a bug! I'll look into adding wiki info on how to adjust that if you want to.

cyoung, 2017-05-24:
@daveholland86 - if it's running slowly then it's working fine. The speed will increase as the temperature does.

cyoung, 2017-05-24:
Looks like frequency varies with duty cycle:

https://raspberrypi.stackexchange.com/questions/4906/control-hardware-pwm-frequency

lukepalmer, 2017-05-24:
There are a couple issues. The built-in balanced PWM mode does odd things
at times. There is also a library incompatibility issue and a weird setup
ordering issue that only shows up after rebooy... Will send a PR once I'm
certain.

cyoung, 2017-05-24:
The PWM duty range also seems to affect the frequency. This is getting a little more complicated than it is worth for the ~8% power savings maximum. From the PI-FAN that I have, it seems like this fan has a very narrow controllable speed range. The SUNON fan is much better in this aspect.

Here are some measurements on a bench power supply.

SUNON
91 mA @ 5.00V
86 mA @ 4.75V
81 mA @ 4.50V
76 mA @ 4.25V
73 mA @ 4.00V
69 mA @ 3.75V
64 mA @ 3.50V
61 mA @ 3.25V
56 mA @ 3.00V
51 mA @ 2.75V
47 mA @ 2.50V
43 mA @ 2.25V
39 mA @ 2.00V
34 mA @ 1.75V
31 mA @ 1.50V
23 mA @ 1.25V
9 mA @ 1.00V (stopped moving)

PI-FAN
138 mA @ 5.00V
128 mA @ 4.75V
117 mA @ 4.50V
109 mA @ 4.25V
103 mA @ 4.00V
94 mA @ 3.75V
87 mA @ 3.50V
83 mA @ 3.25V
76 mA @ 3.00V
70 mA @ 2.75V
64 mA @ 2.50V (stopped moving)
58 mA @ 2.25V
50 mA @ 2.00V
37 mA @ 1.75V

Since most of the time minimal air movement in the case is enough to keep things at a reasonable temperature, consider the minimum power in each of these tests to maintain some airflow.

SUNON: 0.029W
PI-FAN: 0.16W

Max power:

SUNON: 0.46W
PI-FAN: 0.69W

The SUNON fan blows considerably more air than the PI-FAN in any case, so I'm going to conclude that the SUNON is far more efficient. If I've got nothing better to do one day, I'll get out the anemometer and do some actual measurements on this part. But considering that the SUNON fan is worth less than $2 in bulk, I can't see how this decision is anything other than trivial. Send me an email with your address and I'll send you one and a fan controller.

daveholland86, 2017-05-24:
If its supposed to operate like that then I am not worried about mine.  I fly out in Phoenix and it was 103, 110 on the ramp and 125 in the plane and the fan was spinning slow sometimes.  I was worried it would over heat but if its set to not overheat than I wont worry about it.  I am using a 10,000 mah battery and a 6000mah backup and fly maybe 4 hours max, not too worried about battery life.

pp4life, 2017-05-24:
I ran update-stratux-v1.3b2-0678955b01.sh this morning for way more than 10 minutes. It was running for hours this morning and got over 60C, but the fan never came on. I updated to update-stratux-v1.3b2-46233e704c.sh and walked away from my desk. About an hour later I came back to my desk and the fan was running. Temperature was around 50C. 

I rebooted and the fan came on, what seems to be full force, after about 5 minutes. The temperature is about 55C and stabilized at 40C after the fan came on. Doesn't seem like the fan ever slowed down since it came on. I think I understand that's how this update works now. "If doesn't like the PWM, then just turn on to full PWM." 

I see Chris' commits above that change the PWM mode and some other stuff. When a .sh is available, I'll try that.

pp4life, 2017-05-24:
Ha! Just tried the dangerzone update-stratux-v1.3b2-7c33203672.sh. I had my headphones on at work and could hear a tone coming from somewhere step decreasing in frequency. I took off my headphones and it was coming directly from the Stratux, then BAM!, tone stopped and fan turned on what seems to be full blast. I assume that's the, "I give up, just turn the fan on"

lukepalmer, 2017-05-24:
@pp4life there are some known problems here, give it a day or two and I'll
gave something new ready.

pp4life, 2017-05-24:
Don't stress about it or anything. Just trying to give as much feedback as possible on my use case with the [FSY27S05L fan](http://www.ebay.com/itm/Computer-Case-Cooling-Fan-25-x-25-x-7mm-5V-DC-Tablets-PC-Case-CPU-Cooler-x-1-/261508509162). 

Thanks for working through this!

lukepalmer, 2017-05-29:
#595 should be ready for another look now

Joelbear5, 2017-06-02:
Is there anyone looking to implement a PID control solution to the fan PWM? All you need is a target temp and the PID will control the fan at whatever speed is necessary to maintain temperature. Could add a user setting for min PWM duty cycle for those with issues turning on below 30%.

lukepalmer, 2017-06-03:
@Joelbear5 you're the first to mention it, but PID is something we could look at down the road if folks are having thermal issues. It may be more than we need for now.

#595 as-is provides both a configurable target temperature and minimum duty cycle for those who would like to adjust.

D35Bonanza, 2017-06-07:
Can someone post the Kp Ki and Kd values for the PWM controller

Helno, 2017-06-07:
The tuning is going to be really tricky since not only do we have massively different heat loads but also different cases and fans. The delay in sensed temperature vs airflow will also cause problems if you try to have really tight control. 

My initial script left it pretty loose and just ramped things up slowly with 2 degrees of deadband, the slow refresh rate seemed to work really well for this application.

lukepalmer, 2017-06-07:
Are folks having heat problems they'd be willing to describe? There are
some known bugs at HEAD that #595 fixes, and once that is merged we may be
in good shape.

We are actually not doing anything terribly far removed from the original
by @helno, the software is just toughened up a bit and is more configurable.

D35Bonanza, 2017-06-07:
The tuning is going to be really tricky since not only do we have massively different heat loads but also different cases and fans. The delay in sensed temperature vs airflow will also cause problems if you try to have really tight control.
yes I agree
Ignore the heat load and start the fan first. then begin a control method. You cannot get to cold.

Helno, 2017-06-07:
The temperature insensitivity of this whole thing is why a simple temperature control method works well. 

The only things I would have added if I were able to would be to have thresholds to put the  fan at full power or turn it off completely. If it is between those two limits a 5 second refresh time with a 1% output change seems to work just fine.

D35Bonanza, 2017-06-07:
Then make it bang bang control at first. if the temp is high bang on, set a time period say 60sec, then turn it off for 30sec. wait 15sec and measure, start over, until you can determine a control PWM duty cycle while the fan is running at speed.

Helno, 2017-06-07:
Why do bang bang when starting at 50% and moving 1% every 5 seconds works so well.

I think you guys are really overthinking this. We don't need to actually control the temperature at a precise temperature we just want to reduce the fans power consumption when it it not needed. It is more important that the temperature changes slowly than to have it at any specific temperature. 

I tune PID loops for a living and can tell you that PID is a bad idea for this application since the process can change so much from unit to unit and from day to day. Integral only control with limits would work just fine.

lukepalmer, 2017-06-07:
+1 on what @helno says. I am optimistic that we will be in pretty good
shape once #595 gets merged.

D35Bonanza, 2017-06-07:
Because people are saying the fan does not start. each fan has a different startup current. once started you can change to 50% or whatever 
 post your kp ki and kd for integral only control

I myself use conduction cooling and do not use a fan.

lukepalmer, 2017-06-07:
@D35Bonanza agree, there are bugs in the current version at head that cause the fan to not start for some people. Once changes are merged we should be good.

lukepalmer, 2017-06-07:
I just did a side-by-side test between the SunOn MagLev fan that @cyoung recommended and the PI-FAN that comes with some of the cases. I updated #595 with recommended settings for the SunOn fan, as well as suggested settings for the PI-FAN.

If you have the option, the SunOn fan appears to be _much_ better in every way.

D35Bonanza, 2017-06-07:
Luke, I wish you the best of luck. My "avionics fan" runs full speed, squirrel cage blower, there is plenty of air in and around the radio stack. I'm very unclear why a fan is used with Rpi3 and two SDRs, conduction cooling works for me. I may give the fan circuit a try for fun when I get a chance.

D35Bonanza, 2017-06-07:
Chris is correct
"This is getting a little more complicated than it is worth for the ~8% power savings maximum"
http://www.mouser.com/ds/2/256/MAX31740-253424.pdf
This is what is used when needed. No software required.
PCB on linux to make the single layer board. ultra-simple
http://pcb.geda-project.org/

Helno, 2017-06-07:
Not sure what benefit that chip does for us other than replace the software. 

It still needs an external temperature sensor (We have that with the pi cpu) It still needs an external transistor (which we have on the AHRS board)

closed 2017-10-23 by cyoung