cyoung/stratux #600: GPS Disconnected

sgoph, 2017-05-15:
1. Stratux version: 1.3b2
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type: RY836AI

    AHRS
      - [x ] yes
      - [ ] no

    power source: wall power supply

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: No GPS Sats - After updating to 1.3b2, the GPS has stopped working and constantly shows Disconnected. GPS functioned fine with previous 1.3. GPS works when connected to PC using usb; 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-05-15:
Could you go to http://192.168.10.1/ while connected to Stratux, copy the version string/commit at the top of the page, and paste here to confirm, please?

[image]

sgoph, 2017-05-16:
This is what I'm showing - v1.3b2 (95fe19902a)

cyoung, 2017-05-17:
That's the right version - does it do the same after power cycling it (disconnect power source completely, then reconnect)?

westphae, 2017-05-18:
I have a possibly related issue after with v1.3b2 and later commits.  Oftentimes after I restart the stratux process (systemctl restart stratux.service) then GPS shows as disconnected and doesn't reconnect.  Cycling the GPS toggle on the settings page doesn't help.  Stopping and starting the process again will usually restore GPS after one or two tries.  Before the recent changes to gps.co (v1.3b1-2) I never had this issue.

peepsnet, 2017-05-18:
I ma seeing the same with v1.3b2 (069cb9e1fc)

sgoph, 2017-05-18:
Interestingly, after 2 days of it not working, many power cycles, and hours of it running with GPS listed as disconnected, I powered it on today and I did get a 3D fix. It's only listing the G and R labelled Satellites though.

Ergonomicmike, 2017-05-18:
FWIW, I had a problem back around v1.3b1 where the GPS didn't come on line. It took a hard restart of the Stratux. 

Was kind of unusual, but the things been rock solid for about 6 months now.

Haven't had any problems since.

cyoung, 2017-05-18:
@westphae - maybe revert to previous versions and verify that. I assume you're using the GPS over serial? If so, then it's probably the baud rate mismatch issue. When Stratux defaults to a UART GPS, it initially sets the baud rate to 9600. It sends the settings strings then switches to 38400 baud. The GPS should be at 38400 baud at that point. It never sets it back to 9600, so the GPS is sending at 38400 and Stratux is expecting it to be sending at 9600 baud on next restart.

cyoung, 2017-05-24:
Fixed in [latest](http://updates.stratux.me/builds/update-stratux-v1.3b2-46233e704c.sh).

https://github.com/cyoung/stratux/commit/eebb040e1ea2474d45e559f6c3047c3866b5391d

Re-open if further issues.

closed 2017-05-24 by cyoung