cyoung/stratux #602: Fancontrol.py needs to be adjusted

roperf15, 2017-05-19:
1. Stratux version:1.3r2
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - x[ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Foreflight

    EFB platform: (e.g., iOS 9.2)iOS 9.3

    EFB hardware: (e.g., iPad Mini 2) iPad 3

4. Description of your issue:
  The new settings for Fancontrol won't start my fan. It just sets there making a high pitch whine. I expect I need to change the pwm settings that were changed in 1.3r2. 

  Unfortunately I don't mess with the files in stratux very much and especially not since all the great tools were added to the web ui. Where is Fancontrol.py in the file system?  I want to nano into it and try some different pwm settings. I have exhausted my knowledge on Linux file systems looking for it. Thanks in advance

Jim

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-05-19:
Jim, it's installed at `/usr/bin/fancontrol` and it is compiled from `main/fancontrol.go`. You can compile it by just typing `make fancontrol` in the main directory. Make sure you `/usr/bin/fancontrol stop` before you do that. You will need [the latest dev image](https://github.com/cyoung/stratux/releases/download/v1.3b1/stratux-v1.3b1-a6f9aaf9f5-dev.img.zip) and [to install a static wiringPi library](https://github.com/cyoung/stratux/commit/cab2a81c26f214709f8cda941ff8ea19ca4549ba#commitcomment-22152482).

cyoung, 2017-05-24:
@roperf15 - try the latest [here](http://updates.stratux.me/builds/update-stratux-v1.3b2-46233e704c.sh). You may need to give it 10 minutes before the script puts the fan to full speed. What fan are you using?

closed 2017-05-24 by cyoung