cyoung/stratux #604: Is it possible to receive pressure altitude in AHRS report?

pasniak, 2017-05-23:
1. Stratux version:  1.3b2
2. Stratux config:

    SDR
      - [x] single
      - [ ] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [x] yes, InvenSense MPU-9250 and Bosch BMP280 from Amazon. 
      - [ ] no

    power source: Anker

    usb cable: split

3. EFB app and version: Avare

    EFB platform: Android

    EFB hardware: many

4. Description of your issue: I implemented simple AHRS protocol (0x4c) client for Avare Helper.  I would like to add pressure altitude display (29.92, etc...) just like on https://upload.wikimedia.org/wikipedia/en/6/69/G1000_PFD_screenshot.png. How do I go about it?

cyoung, 2017-05-23:
Modifying the 0x4c message isn't ideal since we added that to get into EFBs with already-existing AHRS support. Adding to that may break support on other EFBs.

If you're adding the client from new, I suggest you use something else. Are you able to make a websocket connection to Stratux and parse incoming packets? `ws://192.168.10.1/situation` will stream [this](https://github.com/cyoung/stratux/blob/master/main/gps.go#L58-L92) data every time there is an update.

pasniak, 2017-05-23:
Sure, I implemented code in AvareHelper which works with minimal latency (I believe it is a network broadcast which triggers AHRS display as no messages appear when Stratux is not moving - IMHO it is also efficient from the tablet's power consumption perspective). But I will add a pull every few seconds or so - if I pulled pitch/bank, they might not be that smooth. Thanks!

pasniak, 2017-05-24:
It works.

Since yesterday I could not get it to work :(  I connected to a WebSocket but got no onMessages...  Then I realized no error occurred but the socket was closed with: "draft org.java_websocket.drafts.Draft_10@e7f021 refuses handshake"

The trick was to use:
`
        mWebSocketClient = new WebSocketClient(uri, new Draft_17()) {
`

pasniak, 2017-05-24:
Yet, I now get ~17K/s of data traffic instead of 4K/s when using AvareHelper only...

cyoung, 2017-05-24:
JSON is not the most bandwidth efficient but both ends should be able to handle it. How is the CPU load?

pasniak, 2017-05-25:
Monitors on
1) Avare without ws://192.168.10.1/situation listener (no network because only AvareHelper is connected)  https://i.imgur.com/00tp7EM.png
2) AvareHelper processing AHRS 0x4c reports https://i.imgur.com/mNOnMyg.png
3) Avare with ws://192.168.10.1/situation processing pressureAltitude from onMessage (+getting AHRS from AvareHelper) https://i.imgur.com/FozQcbj.png

cyoung, 2017-05-25:
What is the increase in CPU in #1 attributed to?

kdknigga, 2017-05-25:
These stats are from the Android device?

pasniak, 2017-05-25:
yes, these are Android Studio "Monitors" connected to my Huawei tablet.  I will run more profiling tomorrow. (I used my today's time allocation to hook up webSocket callbacks to the PFD display)

cyoung, 2017-06-06:
@pasniak 
Regarding the original request:

1. Is parsing `ws://192.168.10.1/situation` provide sufficient information for your application? (pressure altitude, etc.)
2. Can this data feed be parsed by your application on the average tablet without increasing CPU/RAM requirements too much?
3. If "no" to either of these questions, what else is needed?

cyoung, 2017-06-09:
https://github.com/cyoung/stratux/blob/master/main/gen_gdl90.go#L277-L283

Pressure altitude can be obtained from the Ownship report, also.

cyoung, 2017-08-04:
Stale ticket, please respond to questions above and reopen if there's still an issue.

closed 2017-08-04 by cyoung