cyoung/stratux #605: Add Ethernet settings

pasniak, 2017-05-25:
When I plug Ethernet cable I cannot see Stratux on my network. It would be great to have it accessible from desktop for debugging or log viewing.

cyoung, 2017-05-25:
`ifup eth0` from the Wi-Fi connection.

pasniak, 2017-05-25:
Where is the file? Yes, I'm familiar with Linux but each time I get to look at Pi's settings, network config is a different file, which looks somewhat cryptic.... + I'd have to do it every time I get a new Stratux img.... +If I f.up I'd have to reinstall img (it's a headless setup)...

cyoung, 2017-05-25:
/etc/network/interfaces

pasniak, 2017-05-26:
I added  ifup eth0  to /etc/network/interfaces
in the second (empty) line (before something with eth0 already)
Now, I do not see the ip nor I can bring up WiFi (Edimax is not even blinking).

pasniak, 2017-05-26:
I had to ssh into stratux and execute `sudo ifup eth0`.
If it is that simple why not add it to the gui? (when we can already run update shell from there...)

peepsnet, 2017-05-26:
This is not a planned way of using stratux. When we look at features they have to make sense in the common way people will use the device.

Your example is one of development level needs. 

Try to imagine an instance where a pilot would need to know the IP of eth0 while flying the plane and looking at traffic...

⁣Sent from BlueMail ​

pasniak, 2017-05-26:
I meant adding this in `http://192.168.10.1/#/settings`: https://i.imgur.com/QRsdvQV.png

ssokol, 2017-05-26:
I've had a number of customers ask for oddball network configurations for
various reasons. I'm beginning to think a Networking tab that controls the
wireless and the wired Ethernet port might be something worthwhile. It
would be hidden by default but would appear using the same trigger as the
Developer tab.

Some of the networking requests I've seen are:

- Wireless client mode (system attempts to join an existing wireless
network, falls back to AP mode if connection fails
- Wireless security (enable WPA2 password on the AP)
- Manual IP destinations for GDL-90 streams
- Ethernet port automatically starts up with DHCP client
- Ethernet port starts up with static IP
- Ethernet port starts up with static IP and DHCP server

The number of these that I've seen has been minimal, so it shouldn't be
part of the "normal" Stratux UI experience.

peepsnet, 2017-05-26:
@ssokol, 
The Idea of that is great but, I believe, there are too many variables in play.

> - Wireless client mode (system attempts to join an existing wireless
> network, falls back to AP mode if connection fails

This, in theory, sounds easy but... 
1) If you connect to an existing network then by default you are not able to get data from 192.168.10.1. You would have to then set the "Static IPs" under "Developer Options" in "Settings". 
The solution I have gone with is adding an EDIMAX dongle and modifying the appropriate files. 
See: https://github.com/cyoung/stratux/blob/master/image/interfaces#L13-L55

> Wireless security (enable WPA2 password on the AP)
1) https://github.com/cyoung/stratux/commit/575d51c1722a61edf8aa16fe511c6b2f8d311879

2) I can try this again to see if I can get it to work again...

> - Manual IP destinations for GDL-90 streams
This exists in Settings->Developers Options-> Static IP's

> - Ethernet port automatically starts up with DHCP client
These all seem like Developer Options. If you want to help develop then I believe you might need the knowledge on how to do these things 

> - Ethernet port starts up with static IP
Seems like the users DHCP server can take care of this. Assign IP by MAC address. This is what I do. 

> - Ethernet port starts up with static IP and DHCP server
Again this seems like DEV stuff. Are there really users plugging their Stratux into a router in the aircraft and allowing all on board to watch the Status page in flight?? Seems to me like this is a custom application that can be accomplished with a WAP and is not up to us to do this. 

Just my thoughts. 

Also as a note to the processes of the first request. 
Workflow as I see it. 
1) When the RPI boots and wlan0 in started the interface is brought up and then stratux-wifi.sh is started(see: https://github.com/cyoung/stratux/blob/master/image/interfaces#L11)

2) This takes the settings we set in dhcpd.conf, isc-dhcp-server, and interfaces and apply the hostapd file to them. 

It seems like the current process would have to change before bringing up the wlan0. 

I guess I could write a script that looks at the Stratux Config file and see if the user wants to connect to a specified AP(imagine: apName=MyPlanesWiFi). If so then move files around to use the appropriate required settings.

It would require rewriting the interfaces and wpa_supplicant.conf on the fly with every boot. 

Just a quick bunch of thoughts for the process. Not FUN! But Possible... I wouldn't shy away from the challenge if there is anough need and someone to build the web interface and modify the associated .GO files as needed. 

Don

pasniak, 2017-05-26:
Yesterday, just executed `sudo ifup eth0`, wouldn't it be sufficient if this was executed from web server? 

When I run it I got both WiFi (Edimax blue light blinking) and Ethernet working.  Just trying to go for low hanging fruit here...  It's not about anything in aircraft but debugging or even getting logs off, which is much more convenient from desktops, typically plugged into routers.

cyoung, 2017-05-26:
Throw it in `/etc/rc.local` (before the last line) so it does it every time. It doesn't matter if you have ethernet plugged in or not. Not sure what I changed but it should come up automatically. Try either that or `systemctl enable dhcpcd` and see if it comes up on ethernet every time you reboot.

peepsnet, 2017-05-27:
I use an app like this on my Android tablet to create a few commands I use regularly 
https://play.google.com/store/apps/details?id=crosien.HotButton

Like "sudo shutdown -h now"

You could do this for the ifup also!!

D35Bonanza, 2017-06-06:
add entry /etc/network/interfaces
allow-hotplug eth0 //interface is active on boot.
before
iface eth0 inet dhcp //my setting

ifup eth0 no longer required.

see man interfaces

bradanlane, 2017-07-05:
Alternate Option:

It's possible to have a Raspberry Pi configured to automatically switch its WiFi between access point and client mode.

This would still require exposing some admin UI, but it would also mean not having to expose the ethernet port or using a wired connection.

jocamero, 2019-02-04:
@cyoung  Any progress on this?  Or is there a tutorial out there somewhere where I can figure out how to modify the network settings so the Stratux connects to my iPad's Personal Hotspot?  This way the iPad would have cellular Internet while on the ground and the Stratux would be able to get software updates without opening the case to expose the Ethernet port.

I think my preference would be on the following order to connect to iPad's Personal Hotspot, home network SSID, and if neither of those are available, then revert to the default AP mode offering up the Stratux network SSID.  Possible?

wa9als, 2019-02-04:
> @cyoung [1] Any progress on this? Or is there a tutorial out there somewhere where I can figure out how to modify the network settings so the Stratux connects to my iPad's Personal Hotspot? This way the iPad would have cellular Internet while on the ground and the Stratux would be able to get software updates without opening the case to expose the Ethernet port.

I think you can get what you want by making your ipad-Stratux connection
static (mine is 192.168.10.18) and give it a fixed DNS as well like
8.8.4.4 and router 192.168.10.1. I do this with my ipad, then the ipad
can still use the Stratux and also connect to my phone hotspot and use
Internet.

 

Links:
------
[1] https://github.com/cyoung
[2] https://github.com/cyoung/stratux/issues/605#issuecomment-460392953
[3]
https://github.com/

HoeckDK, 2022-08-10:
Hi all,

I'm looking for the eth0 DHCP server+static IP. 
My use case is that I have run out of serial in my Open Vario and would like to avoid using wifi. Therefore if the LAN port would act as the WLAN does, it would be sweet, and provide the vario with an IP.

I have tried to modify /etc/network/interfaces, but changes seems to be reverted on next reboot. And I never manage to get an IP other that 127.0.0.