cyoung/stratux #606: Fan does not start in v1.3b2 (95fe19902a)

FlyingSikh, 2017-05-26:
1. Stratux version: v1.3b2 (95fe19902a)
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ X] yes
      - [ ] no
      type:GPYes

    AHRS
      - [X ] yes
      - [ ] no

    power source: 2.4A

    usb cable: 6inch

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
FlyQ
    EFB platform: (e.g., iOS 9.2)
iOS10.2
    EFB hardware: (e.g., iPad Mini 2)
iPad Air2
4. Description of your issue:

Fan does not start, webui temp showing 63.4C. If I downgrade to v1.3b1 and use update-stratux-v1.3b1-a6f9aaf9f5.sh, fan works but I can never lock on GPS satellites. GPS button in GPS/AHRS menu always shows RED. with v1.3b2 (95fe19902a), GPS locks on but the Fan never comes on. Currently at 12:35am the CPU temp is displaying 65degsC indoors. Im afraid, sitting on the panel while flying to Palm Springs tomorrow, it might get fried.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-05-26:
Give the [latest build](http://updates.stratux.me/builds/update-stratux-v1.3b2-506db50fb2.sh) a try.

FlyingSikh, 2017-05-30:
Thanks, tried latest build update-stratux-v1.3b2-506db50fb2.sh but No Joy with this ver. either, fan wont start. waited over 30mins. 
Downgraded to v1.3b1 and use update-stratux-v1.3b1-a6f9aaf9f5.sh and fan is running what seems to be Full Speed holding CPU temp at 41degc C.
By the way, What is the best way to downgrade? Format and reinstall? I couldn't just reload *.sh file.
Also, I have had corruptions a couple of times. What is the best method to power on/off the unit? Please point me to a guide if there is one.
TIA

cyoung, 2017-06-06:
Fixed in #612 with https://github.com/cyoung/stratux/commit/7fd67ccbc17686014158b39d838d2377cc4baaa3.

http://updates.stratux.me/builds/update-stratux-v1.3b2-45e8ea90ac.sh (or later).

closed 2017-06-06 by cyoung