cyoung/stratux #608: FltPlan Go nothing shows

shuffleman319, 2017-05-27:
1. Stratux version: 1.2.2
2. Stratux config: 

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:
Raspberry pi power adapter
    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
FltPlanGo 4.3.5
    EFB platform: (e.g., iOS 9.2)
iOS 9.3.5
    EFB hardware: (e.g., iPad Mini 2)
Ipad 2
4. Description of your issue:
No Planes or Towers are being shown on flight plan go yet there are some shown in the browser interface.

AMNegron, 2017-06-04:
First, any reason you're still on iOS 9 and FPG 4.35? If possible, go to iOS 10 and FPG 4.5.3

Within FPG, under External, ensure the Stratux is selected.
The status screen here provides Towers, Traffic and Data counters. If any of the status icons are not green, report back.

Under Maps, ADS-B, ensure a data stream is selected (on the left side).
For Traffic, select the "gear" icon and for now ensure that "Hide Distant Traffic" is unchecked.

TimTaylorToo, 2017-06-05:
My Stratux works fine with FltPlan Go including AHRS on my iPad Mini 4.

cyoung, 2017-06-06:
Working OK on an iPad Mini 3.

@shuffleman319 - follow @AMNegron's suggestion and reopen if you have further issues.

closed 2017-06-06 by cyoung

groupboard, 2025-06-30:
Note that you also need to ensure the Traffic has a tickbox next to it on the ADS-B pulldown. For some reason it seems to untick itself regularly, which results in no traffic being displayed. FltPlan Go doesn't have the most intuitive UX.