cyoung/stratux #612: v1.3b WingX AHRS Display

tschlabs, 2017-06-03:
1. Stratux version: v.1.3b2 (0a60929cab)
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type: vk-162

    AHRS
      - [x ] yes
      - [ ] no
Stratux AHRS Sensors and Fan Controller / Raspberry Pi MPU9250+BMP280
    power source: batter

    usb cable:

3. EFB app and version:  WingX Pro7 8.6.13.1

    EFB platform: IOS

    EFB hardware: IPAD/IPHONE

4. Description of your issue:

When checking the stratux status page, AHRS functions perfectly. In WingX Pro7, the ahrs data will not display, rather giving a red X. When Checking the stratux status in the external hardware page of wingx, It shows that the embedded AHRS is working (Green)  and Identifies the RY835AI. On previous version of v1.3b the ahrs was working, not now.

TimTaylorToo, 2017-06-04:
Mine works great with iFly but won't work with Wing X Pro 7.

TimTaylorToo, 2017-06-05:
Also works with FltPlan Go, but not with Wing X Pro 7.

TimTaylorToo, 2017-06-05:
Does anybody know how to resolve this?  I'm building a Stratux to sell on eBay and it would be good if Wing X Pro 7 would work with the AHRS.

peepsnet, 2017-06-05:
Tim,
I believe you are not fully understanding the why github is used. Github is for addressing issues that exist in the software and introducing new features to the software. 

If you want to tackle the issue of WingX fell free to start here
https://www.google.com/search?q=golang+training

The way to "resolve" this would be to add the necessary code to the application yourself or wait till someone wants to take the time out of their day, for free(because that is what OPEN SOURCE is), to add it. 

If you are looking for help with your stratux try the reddit group...
https://www.reddit.com/r/stratux/

cyoung, 2017-06-05:
@tschlabs - did you figure out the issue ?

tschlabs, 2017-06-05:
No I have not, sorry about all the confusion I'm new to all this. I'm a NOOB

cyoung, 2017-06-05:
@tschlabs - you didn't do anything wrong. I was able to reproduce this.

Stepping through the releases in http://updates.stratux.me/builds, it looks like a change between
https://github.com/cyoung/stratux/commit/95fe19902a944a1f69ec0aeb4e54ee2bf9f8543b

and

https://github.com/cyoung/stratux/commit/069cb9e1fc617e124ee098df6e648c284289312b

broke WingX support.

tschlabs, 2017-06-05:
I was just going back through it myself. Thank you

cyoung, 2017-06-05:
https://github.com/cyoung/stratux/compare/95fe199...069cb9e#diff-661fb00a6c5b08bcc81eb76962cf9bc2R280

https://github.com/cyoung/stratux/commit/7add14badf536f59cb6b1aa9073a795e893ef984#diff-5fc4419ffcd6649f06a2e1c291e829beR1905

Heading appears to be a required field for WingX, so this change broke the display.

cyoung, 2017-06-05:
Try http://updates.stratux.me/builds/update-stratux-v1.3b2-23921ee384.sh @tschlabs

tschlabs, 2017-06-05:
(http://updates.stratux.me/builds/update-stratux-v1.3b2-23921ee384.sh works perfectly. No AHRS works and fan is cycling as it is suppose to. Thank you. @cyoung

FlyingSikh, 2017-06-06:
I just tried build (http://updates.stratux.me/builds/update-stratux-v1.3b2-23921ee384.sh), fan does not work. I go back to update-stratux-v1.3b2-95fe19902a.sh after formatting and reloading v1.3b2, all is well.
No other build since 95fe19902a.sh has worked for fan on my unit. What am I doing wrong?

cyoung, 2017-06-06:
@FlyingSikh - please define "work". Does the fan spin up on reboot at all? What fan are you using?

FlyingSikh, 2017-06-06:
No sir does not spin on reboot at all. It is a SUNON fan.

cyoung, 2017-06-06:
Interesting. Could you try re-imaging using [v1.3b1](https://github.com/cyoung/stratux/releases/download/v1.3b1/stratux-v1.3b1-a6f9aaf9f5.img.zip) and then upgrading using the above .sh?

FlyingSikh, 2017-06-06:
u mean upgrade straight to 23921ee384.sh?
from 1.3b1 or 1.3b2?

cyoung, 2017-06-06:
v1.3b1 image, then .sh update to 23921ee384.

tschlabs, 2017-06-06:
Whenever I work on mine, I found it best to go back to v1.2r2 and update from there. If the fan wasn't working on a particular update, any updates seemed not to work either so I would start back over. May be doing more work than required but it has worked for me

cyoung, 2017-06-06:
Sounds like an update problem. I think the above update should fix that issue. @FlyingSikh - give [this update](http://updates.stratux.me/builds/update-stratux-v1.3b2-45e8ea90ac.sh) a try (doesn't matter if you re-image or not).

FlyingSikh, 2017-06-06:
ok..here ar emy results..with just stratux-v1.3b1-a6f9aaf9f5-dev.img...fan runs full speed.
With update-stratux-v1.3b2-23921ee384.sh ...fan does not run at all.
With update-stratux-v1.3b2-95fe19902a.sh ..fan runs when needed
With straight from v1.3b1-a6f9aaf9f5-dev.img then update-stratux-v1.3b2-23921ee384 fan spun up at reboot then Off, .then few seconds later spun up for a 5secs then off.
Temp 50.5 Fan kicked in at normal speed. Temp 48.3 Fan stopped.
Temp 50.5 Fan kicked in again at normal speed..Temp 48.3 Fan stopped
Temp 51.5 Fan kicked in again at normal speed..Temp 48.3 Fan stopped

Seems to be working..you are awesome man..!!

Take away for me is that always install new .sh file from a clean image.
Thank you.

FlyingSikh, 2017-06-06:
Thanks update-stratux-v1.3b2-45e8ea90ac.sh working fine too except the fan ON set point is at 52.1C

cyoung, 2017-06-06:
Excellent. The setpoint is the same (50C), it should maintain the temperature "around there". It won't ever be exact.

Thanks to both of you. Looks like we fixed that update bug that was causing the fan controller to not update with the .sh method.

I'll close this now and we can reopen later if we experience further issues with WingX Pro 7. If there are further issues with the fan controller, let's make another issue.

closed 2017-06-06 by cyoung

tschlabs, 2017-06-06:
Thank you!! Keep up the good work