cyoung/stratux #613: Can't Password Protect My Stratux

TimTaylorToo, 2017-06-03:
1. Stratux version:1.3b2
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [ x] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)iFly GPS Beta

    EFB platform: (e.g., iOS 9.2)IOS 10.3.2 and Windows 10

    EFB hardware: (e.g., iPad Mini 2)iPad Mini 4 and Surface Pro 4

4. Description of your issue:Can't password protect my Stratux using Putty.  

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

peepsnet, 2017-06-04:
If you are talking about the new method of simply typing open or secure...

That is not valid yet. I am waiting for the PR #611
https://github.com/cyoung/stratux/pull/611

TimTaylorToo, 2017-06-04:
Is there a way currently to password protect my Stratux??  Thanks.

TimTaylorToo, 2017-06-04:
I was able to do it once before using an earlier version of the Stratux software, but I can't get it to work now.

peepsnet, 2017-06-04:
It was taken out. Somehow it broke. I got it to work again... when the PR is incorporated then you can update and it will be available

TimTaylorToo, 2017-06-04:
OK, thanks.

TimTaylorToo, 2017-06-04:
By the way, the AHRS is working great with iFly GPS beta.

cyoung, 2017-06-04:
@peepsnet - Will get to your PR soon. I seem to remember some issue with getting WPA2 to work? Was that an EDIMAX issue only?

peepsnet, 2017-06-04:
Well. On my pi3 all is good. Remember I also have the edimax dongle so I can connect to my home wifi. It has always been random as to if wlwlan0 was the edimax or onboard. But I have had no issues on my pi 3

TimTaylorToo, 2017-06-04:
I am using a Pi3 with built-in WIFI.  When I used the Putty app to get into the Stratux program, I was able to change the SSID, but the command for adding a password would not work.  As a matter of fact, I think it was gone.

peepsnet, 2017-06-04:
Tim. 
Some time ago I removed the functionality for encrypting the Stratux wifi. There was an issue I could not overcome. I gave it another try recently and seemed to be successful.  

The way this works is I submit the new code to @cyoung and he looks over it and decides if it is "worthy" of being added to the Stratux project. These are called pull requests. 

Since cyoung(Chris) is the guy who started.this whole Stratux thing its only fair he gets a say.

Sit tight and wait for word. If you are getting email notifications from github about this and the pull request #611 you will see a message which says "merged". Merged means my code was... merged with the "master" code.

TimTaylorToo, 2017-06-04:
Thanks.  I wasn't sure if he was asking me a question or not.

TimTaylorToo, 2017-06-04:
The reason I want a password is I don't like leaving file sharing open on my Windows 10 computer.  I want to make my Stratux a password protected private network so I can close down file sharing.  This only became an issue since iFly GPS introduced their Windows 10 beta version.

TimTaylorToo, 2017-06-04:
Problem resolved.  Thanks Chris.

closed 2017-06-04 by TimTaylorToo