cyoung/stratux #614: GPS Accuracy displayed as "zero"

AMNegron, 2017-06-05:
1. Stratux version: 1.3.1
2. Stratux config:
    SDR - [x] dual
    GPS - [x] yes = VK-172
    AHRS - [x] no

    power source: Innergie PocketCell - 3,000 mAh 2.1A
    usb cable: USB2.0 / 28AWG

3. EFB app and version: FlightPlan Go 4.5.3

    EFB platform: iOS 10.3.2

    EFB hardware: iPad Mini

4. Description of your issue:
GPS Accuracy is not being displayed by FPG.
[image]

cyoung, 2017-06-06:
I've submitted this to the FltPlan Go team.

Keeping this open to track it.

D35Bonanza, 2017-06-07:
Same on Avare,, it bounces around 0 or some ft number,. no issue with past stratux releases same number as the stratux webpage.

cyoung, 2017-06-07:
@D35Bonanza, @AMNegron  - which version/build number?

D35Bonanza, 2017-06-07:
stratux-v1.2r2-bcd62e7c00
The Stratux webpage show a number.
 Avare cannot create a GPS Constellation map at times. Often is is blank, however location Lat/Lon is correct. Altitude accuracy is taken as zero from Avare calculations it appears. Sometime is can make a calculation then drops it quickly most times it cannot.

AMNegron, 2017-06-10:
I'm using v1.3b1 (a6f9aaf9f5)

AMNegron, 2017-06-15:
Accuracy error also showing in latest Aerovie. Also of note, Aerovie is showing the negative altitude error. I've advised Aerovie of both, but just wanted to make you aware.

[image]

Helno, 2019-02-01:
Has anyone tried with newer versions of fltplan go or Aerovie to see if these bugs still exist?