cyoung/stratux #615: No WiFi connection to Stratux page after updating to 1.3b2

Leadfoot42, 2017-06-05:
1. Stratux version: 1.3b2

2. Stratux config: 
Dual SDR, GPS, AHRS, running on AC adapter

3. EFB app and version (e.g., WingX Pro7 8.6.2). EFB platform (e.g., iOS 9.2). EFB hardware (e.g., iPad Mini 2).
WingX Pro7 v8.6.13 on Ipad mini4, iOS 10.3.2

4. Description of your issue:
After downloading the v1.3b2.sh file I used the Stratus Settings page to update software with downloaded .sh file (using PC).  Got 'Update Successful' message then Rasberry3 rebooted automatically but after reboot I could not connect to the Stratux status page  on iPad or PC even though I could see the Stratux WiFi network.  Also get no green light on Raspberry and no blinking of red light.

cyoung, 2017-06-05:
Just curious, why are you using an AC adapter?

Leadfoot42, 2017-06-05:
Just setting up the system before installing in aircraft.
I was able to get working again by doing a SD burn of 1.2r2 on PC but this version does not support AHRS.

Leadfoot42, 2017-06-05:
I must have done something wrong when I did update on Stratus Settings page using the 1.3r.sh file.  Anyone know the proper way to update using Settings page of Stratux?

closed 2017-06-05 by cyoung