cyoung/stratux #616: Statux Status Logging in Developer Mode, cannot toggle off

D35Bonanza, 2017-06-06:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [x ] no

    power source: 28 vdc Aircraft. ubec 5.5vdc at 2amp

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
Avare
    EFB platform: (e.g., iOS 9.2)
Android
    EFB hardware: (e.g., iPad Mini 2)
Nexus
4. Description of your issue:
Status Logging can be toggled on/.off in Developers Mode for StaticIps use on local LAN eth0 routed to Internet. The APD is not being used on wlan0

Gen_gdl90.go has been commented and make, stxstop, make install, Stxstart shows no status logging being logged for developers mode on Developers window Stratux Webpage.

		//FIXME: Only do this if data logging is enabled.
	//initDataLog()

// Start printing stats periodically to the logfiles.
	//go printStats()

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

D35Bonanza, 2017-06-06:
//FIXME is current for gen_gdl90.go

cyoung, 2018-01-17:
Good point. We might just route the normal logging to `/dev/null` unless enabled.

closed 2018-01-17 by cyoung