cyoung/stratux #617: Auto Power On possible?

FlyingSikh, 2017-06-09:
1. Stratux version:1.3b2
2. Stratux config: 

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [ x] yes
      - [ ] no

    power source: EasyAcc 6000mAh 

    usb cable: supplied

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
FlyQ latest IPad Air2

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: is it possible to add Auto Power On when battery pack is charging or senses power being supplied. I'm planning to remote mount with charging cable permanently attached so I never have to touch it. Powering off by WebUI is fine.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

peepsnet, 2017-06-09:
This would seem to be more a feature of the battery pack...

The other option would be to make your own case that holds the batteries and create a circuit that can do this... a power switching control unit

cyoung, 2017-06-09:
So you will have ship power -> pass-through charging battery pack -> stratux.

It's certainly possible but not so much by software implementation.

FlyingSikh, 2017-06-09:
Ok, How does Stratus do it?

Nokomis449, 2017-06-10:
My solution was to buy a cheap IR relay and put it in a small Radio Shack project box.  Ship's USB power plugs into the input side of the relay, and the Stratux power cord is plugged into the output of the relay.  The Stratux sits on the hatrack in back and the handheld remote allows me to turn the Stratux on & off from the pilot's seat.

dmschlab, 2017-06-14:
>Ok, How does Stratus do it?

They designed their hardware that way

cyoung, 2017-08-04:
Hardware mod, not something we can easily do in the software now. Good idea.

closed 2017-08-04 by cyoung