cyoung/stratux #618: Data on fore flight

dmiddlet26, 2017-06-10:
1. Stratux version: V1.0r1
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual X

    GPS
      - [ ] yes
      - [ ] no. X
      type:

    AHRS
      - [ ] yes
      - [ ] no x

    power source: battery pack

    usb cable: YEs 

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Foreflight   V9.1.3

    EFB platform: (e.g., iOS 9.2) iOS  10.3.2

    EFB hardware: (e.g., iPad Mini 2) iPad mini 2

4. Description of your issue:   Issue just started with the last iOS update. So when I bring up the stratux app on the iPad it shows connected and when I click on the traffic tab it bring up a list of traffic and is updating fine.  When I start fore flight  it connects to the stratux and shows connected, but never displays traffic 
Help 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-06-10:
Verify the [traffic layer](http://sportysnetwork.com/ipad/wp-content/blogs.dir/8/files/2014/05/traffic-layer.jpg) is enabled:

dmiddlet26, 2017-06-10:
I fell silly... the one thing I did not even look at. working perfect
Thanks again

Dave

closed 2017-08-04 by cyoung