cyoung/stratux #621: [Feature Request] Usb Ports Power

eagsc7, 2017-06-17:
1. Stratux version: 1.2r2
2. Stratux config: Default

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ ] yes
      - [X ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no
    Motherboard: Raspberry Pi 3 model B v1.2
    power source: Aircraft via 2.1a power plugin 
    usb cable: 10' 

3. EFB app and version: ForeFlight, Updated yesterday...

    EFB platform: iOS 9.2

    EFB hardware:iPad Mini 2

4. Description of your issue:

I'm looking to consolidate my wires in the cockpit.  Currently, using the two lower ports for the USB Antennas, and Top Left for .5a power to my X-Naut iPad Cooler.  What I WANT to do is with the 4th USB Port (Top Left, Furthest from Ethernet port) have 2.1a Output to keep the iPad charged.  Not sure if this is possible, but if there's a setting within the system, that'd be GREAT!!!

Thank-you

Ben

eagsc7, 2017-06-17:
Secondary use COULD be to port the GPS info from the iPad TO Stratux via the plugin, But that could get complicated with iOS

peepsnet, 2017-06-18:
Short Answer... Not possible... 

Long Answer...  Even if it was possible the amount of power you would need to put into the PI would be greater than some of the circuitry could handle. The Pi itself take as much as 2A to run Stratux alone. If it was possible to even pass over 2A to the USB port it would require upwards of a 5A power supply. 

I believe I am pretty close to the appropriate explanation on the topic. 

Don

cyoung, 2017-10-23:
I agree with @peepsnet on this, but it doesn't take 2A to run Stratux (more like 0.95A).

But running iPad charge current through the Pi isn't a good idea. The port will also probably not allow for iPad high current charging for other reasons.

closed 2017-10-23 by cyoung