cyoung/stratux #623: GPS UART Initialization Problem due to Baud Rate mismatch

jake-b, 2017-06-20:
1. Stratux version: v1.3b2-46233e704c
2. Stratux config:  Dual SDRs, RY836AI (UART), AHRS
    power source:  RasPi Wall Adapter

3. EFB app and version: n/a - web interface 

4. Description of your issue:

As far as I can tell, Stratux (in gps.go) assumes that the GPS will always power up at 9600.  I have found two cases where this not the case:

1.  Upon reboot.  Since power is never disconnected from the RY836, it remains at 38400.  When stratus comes back up, it cannot initialize the GPS at 9600.  From the GPS's perspective, a bunch of noise is sent from the host.  I think this causes gpsSerialReader() to fail, and the next attempt to initialize the GPS at 9600-- which fails again, etc

2. After short power downs.  A graceful shutdown of the RPi and removal of power for a 10 count, and reconnecting power.  The GPS comes back up at 38400.  Perhaps there are some capacitors in there or something that hold the settings.   I could see this happening in flight (temporary interruption in power) and it would be bad if it caused you to lose GPS.  (Even though Stratux isn't your primary navigation aid, right? hehe)

roelink, 2017-07-02:
I have exactly the same problem. Reproducible both on a RPI 2 and 3 with the RY836. My alternate setup with a separate GPS and AHRS works flawlessly. Unstable on 1.3b2 and b.3. Earlier versions are stable. tty shows 38400 when functioning, 9600 when not functioning.

peepsnet, 2017-07-06:
I am seeing the same issues.

When the webUI shows the GPS as disconnected I am able to `screen /dev/ttyAMA0/ 38400` and get good data returned. 

Is there some kind of force/check we could do to force the reset of the ry836ai or force stratux to switch to 38400 and test??

boa888, 2017-08-23:
This problem still exists. 
A few days ago @westphae informed me the RY386AI is not working via UART and that it is a known issue. I know westphae and others are working on the problem. Is there any expected date for the fix? Where should I check for updates on the fix?

cyoung, 2018-01-17:
Not sure what is needed to fix this and nobody has stepped up -- it concerns unsupported hardware. Will reopen if anybody wants to take the lead on fixing it.

closed 2018-01-17 by cyoung

roelink, 2018-01-17:
I gave up and I’m using that GPS module in my carPI Navit setup. Works fine there.

Henk

jake-b, 2018-01-17:
I don't have a build environment capable to really make the changes myself, nor the real interest in learning golang to any extent.  I had hoped that the serial port logic could be made more flexible in trying different baud rates when there is a failure to initialize.  I would suspect that this is not specific to the RY83xAI, but any UART-connected module.

But if no one's interested or able to provide such a fix, then I understand the desire to clean up the issues list.