cyoung/stratux #626: Feature Request - Client mode WiFi

moto501, 2017-06-25:
1. Stratux version: All
2. Stratux config:

    SDR
      - [x] single
      - [ ] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [x] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I know this issue has been raised previously, however I didnt see a resolution. Client WiFi is a must. In my usecase, the iPad does not have a sim and relies on WiFi acess only. I carry a portable wifi hotspot in the plane which all wifi devices connect to. With Stratux being AP mode only, it means that the iPad must now disconnect from the internet in order to work with Stratux. If Stratix was client mode, it could participate in the general WLAN. If this is not a feature request, then the process for how to change it manually would be appreciated. I see that there is some limitation on how stratic sends to clients by looking at the DHCP leases, but im sure there is a better way.

Thoughts?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2017-06-25:
This is already possible.

You will need to edit the interfaces file to connect the pi to your network.

Then in developer mode you can enter ip's to tell stratux which ip's to send data to.

moto501, 2017-06-25:
Thanks, thats great to hear. I will give that a go. I will see if we can send to the broadcast address, as having to go into developer mode to change the IP of the client is not a practical solution. However, its an ok workaround for now.

closed 2017-07-02 by moto501