cyoung/stratux #628: could not determine kind of name for C.rtlsdr_set_bias_tee

garyemiller, 2017-06-27:
1. Stratux version:

Today's git head

2. Stratux config:

None, can't get past make.

4. Description of your issue:

```
# git clone https://github.com/cyoung/stratux.git
# cd stratux
# make
[...]

github.com/ziutek/mymysql (download)
go build -ldflags "-X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`" -p 4 main/gen_gdl90.go main/traffic.go main/gps.go main/network.go main/managementinterface.go main/sdr.go main/ping.go main/uibroadcast.go main/monotonic.go main/datalog.go main/equations.go main/cputemp.go
# github.com/jpoirier/gortlsdr
could not determine kind of name for C.rtlsdr_set_bias_tee
make[1]: *** [Makefile:18: xgen_gdl90] Error 2
make[1]: Leaving directory '/u1/src/rtl-sdr/stratux'
make: *** [Makefile:14: all] Error 2
#
```

garyemiller, 2017-06-27:
Log of my running make.

[build.txt](https://github.com/cyoung/stratux/files/1103990/build.txt)

cyoung, 2017-06-27:
Hm, looks like recent changes on github.com/jpoirier/gortlsdr may not compile.
Will take a look later. @jpoirier ?

jpoirier, 2017-06-27:
@cyoung  
I just double checked and get a successful build via :

> cd ~/tmp
> git clone git@github.com:jpoirier/librtlsdr.git
> cd librtlsdr
> mkdir build; cd build
> cmake ../
> sudo make install
> sudo ldconfig
> rm -rf $GOPATH/github.com/jpoirier/gortlsdr
> go get github.com/jpoirier/gortlsdr

garyemiller, 2017-06-27:
Hmm.  I was using package net-wireless/rtl-sdr-0.5.3_p20150730 on gentoo.  When I remove that and pull from the librtlsdr git it works for me.

So that solves my problem.  I'll create a Gentoo bug for this.

garyemiller, 2017-06-27:
Hmm, gentoo installs steve-m/librtlsdr, but you had me install  jpoirier/librtlsdr.git

Any way to get Stratux to work with either librtlsdr?

jpoirier, 2017-06-27:
bias tee is in latest master branch at steve-m/librtlsdr.git so gortlsdr should compile without error provided it was installed from source but for stratux you should be using jpoirier/librtlsdr.git

cyoung, 2017-06-27:
I'll leave this open - we should probably add something to the build scripts or Makefile that checks the base librtlsdr version.

jpoirier, 2017-06-27:
fyi, latest releases:     jpoirier/librtlsdr v0.5.5 <---> jpoirier/gortlsdr
2.10.0

jovrstra, 2018-04-22:
I am trying to compile Stratux and am running into this issue...  Checking out the jpoirier git does not seem to work:
```
root@raspberrypi:~/tmp# git clone git@github.com:jpoirier/librtlsdr.git
Cloning into 'librtlsdr'...
Permission denied (publickey).
fatal: Could not read from remote repository.
Please make sure you have the correct access rights
and the repository exists.
```
Any way around this?
(I'm building on a Raspbian Stretch)
Thanks!

garyemiller, 2018-04-22:
Your URLgit clone git@github.com:jpoirier/librtlsdr.git  needs a password.  It is the URL for developers.  Use the public URL: git clone http@github.com:jpoirier/librtlsdr.git

jovrstra, 2018-04-25:
Thanks! I overlooked that... The git-command that worked is:
`git clone https://github.com/jpoirier/librtlsdr.git`
(still did not finish it all as there were other issues, on my side, so they don't belong in this thread :))

scottbyrns, 2018-06-25:
I ended up having to pull from source after removing using `sudo apt-get purge librtlsdr0 librtlsdr-dev gr-osmosdr` and building as described here above by @jpoirier 

Took some effort to clean up existing elements of SDR installs, it seemed like these instructions didn't work but after some effort, these solutions are correct.

Ran `sudo apt-get install  gqrx-sdr` before `sudo make install` for librtlsdr