cyoung/stratux #629: main/fancontrol.go:15:23: fatal error: wiringPi.h: No such file or directory

garyemiller, 2017-06-27:
1. Stratux version:

Today's git head.

2. Stratux config:

Not configured, can't build.

4. Description of your issue:

```
go build -ldflags "-extldflags -static -X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`" -p 4 main/fancontrol.go main/equations.go main/cputemp.go
# command-line-arguments
main/fancontrol.go:15:23: fatal error: wiringPi.h: No such file or directory
 // #include <wiringPi.h>
                       ^
compilation terminated.
make[1]: *** [Makefile:22: fancontrol] Error 2
make[1]: Leaving directory '/u1/src/rtl-sdr/stratux'
make: *** [Makefile:14: all] Error 2
```

I don't know where to get wiringPi.h, and since I am not even on a RaspberryPi what is the point?

I will eventually install on about a dozen RasPi's at the local EAA, but most of them will not be adding a fan, so why compile?

Build.log: 
[build.txt](https://github.com/cyoung/stratux/files/1104207/build.txt)

cyoung, 2017-06-27:
You're not using a Raspberry Pi - what is your build environment?

garyemiller, 2017-06-27:
I have several build environments, but I am mostly on Gentoo, plus a number of Debian variants.
Soon I would expect someone to want to port to Andoid as well.

For hardware, I have RasPi B's, RasPi2B's, RasPi 3B's, Odroid C1, Orange Pi PC 2, and various Intel.

I'm going to start on Intel, because it is a faster and more pleasant build environment,  Then move on to the others.  Gentoo runs on all my hardware with very minimal differences, so once I get one going the rest of the hardware is almost done.

I could not find a document that lists needed packages.  Is the wiringPi.h that you use from: http://wiringpi.com/ ?

Since most builds will not have a fan, is there a way to make fancontrol.c optional?  I actually heat my Rasperry Pi's up to 67𝆩C because they work best for my purposes (unrelated to stratux) at that temperature.  I hold the Pi temps to 0.1𝆩C with external apparatus.

Normally I'd just dig in and send you patches, but I'm new to go and you don't have a config system yet.

garyemiller, 2017-06-27:
I installed this version of wiringPi: git://git.drogon.net/wiringPi

I had to install the static wiringPi libs.

with that, I have stratux built on Gentoo/Intel

I suggest adding a file, like the attached INSTALL.txt, to help the next builders.

[INSTALL.txt](https://github.com/cyoung/stratux/files/1106630/INSTALL.txt)

cyoung, 2017-06-27:
So you got this to work on amd64 Gentoo? Will wiringPi compile on there? The "-dev" images have all the tools to build.

garyemiller, 2017-06-27:
A tweaked INSTALL.txt, to mention that stratux also needs installing:

[INSTALL.txt](https://github.com/cyoung/stratux/files/1106636/INSTALL.txt)

Is there a way to NOT install in /usr/bin?  User created binaries should always be in /usr/local/bin/.  Only distros should install in /usr/bin/

From Wikipedia:

"Most of the time /usr/local/ is used for installing software/data that are not part of the standard operating system distribution (in such case, /usr/ would only contain software/data that are part of the standard operating system distribution)."

I do not quite have this working, next error:

```
spidey stratux # make install
cp -f gen_gdl90 /usr/bin/gen_gdl90
chmod 755 /usr/bin/gen_gdl90
cp -f fancontrol /usr/bin/fancontrol
chmod 755 /usr/bin/fancontrol
/usr/bin/fancontrol remove
Removing cooling fan speed control based on CPU temperature:			[  OK  ]
/usr/bin/fancontrol install
Install cooling fan speed control based on CPU temperature:			[  OK  ]
cp image/10-stratux.rules /etc/udev/rules.d/10-stratux.rules
cp image/99-uavionix.rules /etc/udev/rules.d/99-uavionix.rules
rm -f /etc/init.d/stratux
cp __lib__systemd__system__stratux.service /lib/systemd/system/stratux.service
cp: cannot create regular file '/lib/systemd/system/stratux.service': No such file or directory
make: *** [Makefile:49: install] Error 1
```

Gentoo almost never has systemd.  systemd will stay out of Gentoo for a long time, systemd is a major PITA.

So autodetecting if the system is cursed with systemd before installing the systemd cruft would be nice.
Or at least not fail the whole make.

I made the systemd stuff optional in my Makefile and now I have a completed build and install.  Not tested yet.  Here is that Makefile:

[Makefile.txt](https://github.com/cyoung/stratux/files/1106671/Makefile.txt)

cyoung, 2017-06-27:
We've focused mostly on using the Raspberry Pi/Raspbian build environment. There is https://github.com/jpoirier/stratux-setup which would probably benefit from additions like that.

garyemiller, 2017-06-27:
cyoung: Will wiringPi compile on there?

Yes, wiringPi compiles with zero changes.  This allows the stratux fancontrol to compile, and the buid to finish.

Also,  git@github.com:jpoirier/librtlsdr.git builds, installs, and runs without changes on Gentoo.

I'm working up to testing stratux by testing the peieces first.

garyemiller, 2017-06-27:
cyoung: We've focused mostly on using the Raspberry Pi/Raspbian build environment. 

Well, so am I.  The Gentoo build environment on Intel is 99.99% identical to the gentoo build environment on the Raspberry Pi.  As I get these bits to build on Intel then it also works for me on RasPi.

garyemiller, 2017-06-27:
"There is https://github.com/jpoirier/stratux-setup which would probably benefit from additions like that."

Which says right at the top: 

```
There is https://github.com/jpoirier/stratux-setup which would probably benefit from additions like that.
```

Seems to me the INSTALL is so interconnected with the git commit that they should be in the same git repo so they can track changes together.

garyemiller, 2017-06-27:
I have gen_sdl190 running.  I did need one tweak to remove support for the embd modules that is old, unmaintained, and not useful on Intel.  I just made initI2CSensors() always return nil.  And set IP 192.168.10.1/24 on a local interface.  No need for any systemd stuff.

Only thing left for me, on Intel, is to get my GPS to work.

Is there any reason you do not just use gpsd and have all the GPS models in the world supported at minimal effort in your code?

Here is a git diff or my 4 simple changes to get this to compile.  The first three could go, unchanged, in your git head:
[diff.txt](https://github.com/cyoung/stratux/files/1107023/diff.txt)

garyemiller, 2017-06-27:
Getting one of my GPS to work was simple:

```
ln -s /dev/ttyUSB0 /dev/ublox6
```

Now gen_gdl190 works just dandy on my laptop.  I'll show this off at our next EAA meeting.

I must say the GPS config code is 'interesting'.  Functional, but not gonna support many GPS types or options...

So, with just the simple patch above, and the proper prerequisites installed, gen_gdl190 seems to work just fine on Gentoo/Intel.  Please consider my simple changes for your git head so others have an easier time of it.  I'd be happy to shape the patches to your wishes.

cyoung, 2017-08-04:
Looks like you solved the wiringPi issue. gpsd query in #649.

closed 2017-08-04 by cyoung