cyoung/stratux #630: HTML errors

garyemiller, 2017-06-27:
https://validator.w3.org/ finds html warnings and errors in /var/www/plates.index.html

See this screenshot of validator output:

[image]

bradanlane, 2017-07-05:
It looks like that validator is not aware of several things including:

- Stratus Admin Web UI is a mobile web application
- JavaScript frameworks add their own attributes to HTML elements
- some validator warnings are actually in conflict with W3 media-query specifications

A few of the validator messages suggest an opportunity to clean up some HTML/CSS as the specification has evolved over the past 2 years. At the same time, not all browsers are up-to-date on the specification so changes may do as much harm as good.

JohnOCFII, 2017-07-05:
It looks like the HTML evaluator hasn't been updated in about 4 years either.

bradanlane, 2017-07-05:
Good catch John. That would explain both the erroneous "viewport" messages and the "media query" messages. It may also be why the validator is flagging all of the framework attributes on HTML elements.

If this is an important issue, a better validator should be the benchmark.

garyemiller, 2017-07-05:
There is no such thing as 'mobile html' there is only 'valid html.  The way the Stratux JS is done is counter to the HTML specification.

bradanlane, 2017-07-05:
Correct. A "mobile web app" is written with standard HTML, CSS, and optionally JavaScript. 

The "Stratux JS" uses an open source industry JavaScript framework. The framework was created and open sourced by Google.

garyemiller, 2017-07-05:
'industry standard'?  Can you point me to an RFC, or other similar standard?

I am familiar with the 'HTML Standard', provided by the w3c.org.  I'd be happy to point out specifics how Startux conflicts with the published international HTML Standard.

bradanlane, 2017-07-05:
Just like I never stated "mobile HTML", I didn't state "industry standard".

garyemiller, 2017-07-05:
So we agree the Stratux code does not comply with the HTML industry standard?

bradanlane, 2017-07-05:
Agree to disagree :-)

"Industry standards ... the generally accepted requirements followed by the members of an industry."

garyemiller, 2017-07-05:
So w3c.org is not the consensus of the members of the industry?  Must be fun down in your rabbit hole.

If you don't care to get it right, then I'm not gonna...

bradanlane, 2017-07-05:
The standards bodies are comprised of industry leaders and thus there is a lag between industry advancements and standards bodies.

garyemiller, 2017-07-05:
And the stratux html  appears to lag behind even more.  It still has XHTML cruft that was removed years ago from the HTML standard.

JohnOCFII, 2017-07-05:
I'm sure a working Pull Request would be graciously accepted.

bradanlane, 2017-07-05:
(consolidated prior comments)

"Industry standards ... the generally accepted requirements followed by the members of an industry."

_The validator website generates validation errors._ :-)

If contributors feel these validation messages are important, fork the project, make the corrections, and submit a pull request.

garyemiller, 2017-07-05:
"The validator website generates validation errors. :-)"

Uh, no:  https://validator.w3.org/check?uri=https%3A%2F%2Fvalidator.w3.org&charset=%28detect+automatically%29&doctype=Inline&group=0&verbose=1&user-agent=W3C_Validator%2F1.3+http%3A%2F%2Fvalidator.w3.org%2Fservices

bradanlane, 2017-07-05:
I stand corrected. I pasted the wrong URL. I pasted the github URL.

Ergonomicmike, 2017-07-08:
I checked a branched settings.html on both checkers (including the /nu) and they both report errors. I don't mind noodling around to try to reduce the reported errors. But to do that, I need to know what doctype the Stratux wants to use. (And this might already be declared somewhere ... I seem to recall that there was a master webpage for the Stratux UI, and then html was swapped in and out of it.)

Ergonomicmike, 2017-07-08:
So, the first place to start cleaning up - if, in fact, it is necessary, and if the team wants to do it - is with the index.html page. For if it's not right, none of the sub pages that script into it will be either. I remember a year or two ago that the appcache thing was done for good reason. I don't know anything about 'service workers.' This sounds like it might be more than just HTML clean up.

On some of the other error reports, it seems like maybe it's as simple as hyphens aren't allowed in div labels? Even if it's as simple as that, this could become an quite a project, trying to find and retie loose ends.