cyoung/stratux #632: Change SSID hostapd R3b3

737mechdude, 2017-07-06:
1. Stratux version:R3b3
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual  Yes

    GPS
      - [ ] yes yes
      - [ ] no
      type:

    AHRS
      - [ ] yes yes
      - [ ] no

    power source: Battery

    usb cable: 

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Wing X Pro 7

    EFB platform: (e.g., iOS 9.2)   ios 10.32

    EFB hardware: (e.g., iPad Mini 2) Ipad  Mini 4

4. Description of your issue:  Following the steps from here https://github.com/cyoung/stratux/wiki/Change-SSID-(Wifi-Network-Name)-of-your-Stratux  I'm confused because I get "access denied" message.  Also I noticed -s is not an option on the hostapd manager help list of options?  So is this just a beta thing, ie not yet added or won't be?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

peepsnet, 2017-07-06:
you need root access  type `sudo su -`

also 

https://github.com/cyoung/stratux/wiki/Change-SSID,-Channel,-or-Encryption

737mechdude, 2017-07-06:
So I SSH with TERMIUS and I typed hostapd_manager.sh -s stratux-737 and I got a message "invalid option "s"  "  I have a pi 3b  Beta 3b3 of stratux version, I found trying /etc/hostapd manager/hostapd manager.conf and that's where I get Access denied.

peepsnet, 2017-07-06:
What version are you running?

From the webUI on the Status screen there is a version at the top. 

Don

737mechdude, 2017-07-06:
V1.3b3 (74909ada85)

peepsnet, 2017-07-06:
So I want to make sure that you are running it as administrator and what about expanding your file system have you done that

peepsnet, 2017-07-06:
Also try
`ls -lah /etc/hostapd`
Paste results

And

`nano /etc/hostpad/hostapd.user`
Paste results
Then ctrl+X to quit

737mechdude, 2017-07-06:
So not to confuse, I think what I need to do is actually edit the file with a text editor?  I’m only trying to type this stuff in with Termius.  

I’m not expanding file system.  I’m just trying what was on the github forums.  So I need to expand and edit the file?

peepsnet, 2017-07-06:
I'm trying to find doubt where the error is coming from. The only the script does is modify /etc/hostapd/hostapd.user

I am looking to see if it is writable

737mechdude, 2017-07-06:
1s -1ah /etc/hostapd   -bash 1s:  command not found

 

nano /etc/hostpad/hostapd.user   

 

ssid=stratux

channel=1

 

auth algs=1

wpa=3

wpa passphrase=SquawkdirtyToMe!

wpa key mgmt.=WPA-PSK

wpa pairwise=TKIP

wpa pairwise=CCMP

peepsnet, 2017-07-07:
That was a letter L not a 1  

It looks like you were able to secure your wireless network. 

I just go home and was able to successfully change my SSID without issue. 

So I am going to start over with your issue. 

This is the process you should go thgough:
1) SSH into stratux
2) get root with `sudo su -`
3) run the  script by syping `hostapd_manager.sh` and you should see the options( -s is there on my side)
4) type `hostapd_manager.sh -s stratux-N12345`
5) type `y` to confirm the settings
6) if you are connected to your stratux through the 192.168.10.1 interface then you will be disconnected.
7) wait till you see your NEW SSID and connect

if any of this fails please tell me at which step it failed...

Don

closed 2017-08-04 by cyoung