cyoung/stratux #633: GPS Button stays RED

FlyingSikh, 2017-07-07:
1.Stratux version: v1.3b2 ( 95fe199 )

2.Stratux config:

SDR
 single
◦[X ] dual

GPS
◦[ X] yes
 no
 type:GPYes

AHRS
◦[X ] yes
 no

power source: 2.4A

usb cable: 6inch

3.EFB app and version: (e.g., WingX Pro7 8.6.2)
 FlyQ
 EFB platform: (e.g., iOS 9.2)
 iOS10.2
 EFB hardware: (e.g., iPad Mini 2)
 iPad Air2

4.Description of your issue:
GPS Button staying RED even though GPS are seen and resolved.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

FlyingSikh, 2017-07-07:
[image]

cyoung, 2017-07-07:
Which is the "GPS button"?

FlyingSikh, 2017-07-07:
sorry, one under GPS/AHRS.

FlyingSikh, 2017-07-10:
has anyone seen this?
What could be causing it? I have tried formatting and reloading the SD Card. Shall I try a new GPS?
TIA

Ergonomicmike, 2017-07-10:
Perhaps unique to the GPSYes? I have a VK-162 and the red/green light under the web UI's AHRS works just fine for me.

cyoung, 2017-08-18:
I think you're referring to the "GPS/AHRS" page.

Your logs show that the GPS is not locking.

737mechdude, 2017-08-25:
I can confirm I had the same issue, it was the GPS.  Replaced and all is great again.

cyoung, 2017-10-23:
@FlyingSikh - sounds like GPS wasn't getting a lock. Try with the latest and re-open if the issue persists.

closed 2017-10-23 by cyoung