cyoung/stratux #639: Wifi is not broadcasting

737j, 2017-07-17:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

737j, 2017-07-17:
Sorry, I'm not the most technically savvy person... I followed the instructions online and have been using the Stratux for about 3 months. Out of the blue today, it no longer seems to broadcast wifi. Any thoughts?

pasniak, 2017-07-17:
Does the wifi dongle light up?
Are you using the same fuly charged power pack or charger with the same USB cable?

cyoung, 2017-08-18:
@737j - might be a power issue as @pasniak pointed out. Closing the ticket since it has been inactive for a while, re-open it if you're still having issues.

closed 2017-08-18 by cyoung