cyoung/stratux #642: "ADS-B Disconnected" Message In Foreflight

pmaag, 2017-07-18:
1. Stratux version: 1.2r2
2. Stratux config: 

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [ ] yes
      - [X] no
      type:

    AHRS
      - [ ] yes
      - [X] no

    power source: Anker Astro E5 G2

    usb cable: Anker Powerline+ 3ft

3. EFB app and version: (e.g., WingX Pro7 8.6.2)  Foreflight 9.2.1 (12976)

    EFB platform: (e.g., iOS 9.2)  10.3.2 (14F89)

    EFB hardware: (e.g., iPad Mini 2)  iPad Mini 2+Cellular

4. Description of your issue:

When airborne Foreflight will randomly display "ADS-B Disconnected" message while removing all traffic targets from the map view.  This occurs now at least once per flight, usually more frequently.

I have done quite a bit of testing on the ground within sight of a tower.  On the ground Foreflight will not show the "ADS-B Disconnected" error, but does randomly drop all targets from the screen for ~10 seconds, however the iPad's wifi signal never drops.  This seems to occur with high target counts (~5000 1090 messages/min, and ~400 UAT messages/min).  I do not have "hide distant targets" turned on in Foreflight.

I enabled logging within the stratux interface and no errors are reported during the time targets drop on the Foreflight map within stratux.log.  I also had another client connected at the same time as a drop and did not notice any drop from that client into the Stratux WebUI.

I have opened a case with Foreflight to get more information on what the "ADS-B Disconnected" error actually means, and whether or not it is only displayed when airborne (I am guessing it is).

Has anyone else seen the same error message/traffic dropping within their Foreflight maps?  Any ideas on how to troubleshoot?

This seemed to have started around the 1.2r2 upgrade/the new Foreflight upgrades, really hard to pin it down to an exact date.

Thanks!

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

jpuetthoff, 2017-07-18:
Yes, same set-up. Several times, about every 20-30 minutes, ADS-B disconnected message. Wifi connection remains between Stratux and Ipad. Device ADS-B is gone under foreflight devices. Updated to newest revision for stratux to see if it fixes it.

pmaag, 2017-07-23:
I have contacted other users in my area and they are also having this same problem, with varying frequency.

I am able to reproduce the issue on the ground within reception of a tower.  I have a case opened with Foreflight on the issue as well, here is what I am seeing when the issue occurs:

1) Nothing reported in stratux.log during the time of disconnects, or any errors in the OS logs (/var/log/messages) on the Pi.

2) On the ground Foreflight does not display the "ADS-B Disconnected" error, however in the Foreflight logfiles we see the following:

1792 07-23-2017 19:27:06Z |  ~appStatus~ | -[ForeFlightAppMonitor log:includeInCrashLog:] 188 PANEL_CONNECTION_STATE_CHANGED: ADS-B disconnected
1795 07-23-2017 19:27:06Z |  ~appStatus~ | -[ForeFlightAppMonitor log:includeInCrashLog:] 188 PANEL_CONNECTION_STATE_CHANGED: No devices connected.
1798 07-23-2017 19:27:23Z |  ~appStatus~ | -[ForeFlightAppMonitor log:includeInCrashLog:] 188 PANEL_CONNECTION_STATE_CHANGED: ADS-B connected
1801 07-23-2017 19:27:23Z |  ~appStatus~ | -[ForeFlightAppMonitor log:includeInCrashLog:] 188 PANEL_CONNECTION_STATE_CHANGED: ADS-B is connected

So Foreflight is definitely seeing a drop in ADS-B data.  During this time all traffic targets are removed from Foreflight's screen, the "ADS-B: 1 tower" message in the top left disappears.  The wifi icon in the top left never disappears, and other clients connected to the stratux wifi also do not drop.

3) The GDL90 logfile from Foreflight also shows a drop in data:

1500838018.517578,7e14132b8074201c37a89fd11089661070238c0075740000000000000001357e7e1410a9bc82204464a88cf1114999091003ff0165614e373236325000584f7e

1500838043.644208,7e1e00000000acbd03408953512da627d912c806c02f000000000000000000000000000000000000000000000000000000000000000000000000000000000000

Anyone have any ideas on what else to look for during this issue?  I can pretty easily replicate it after being connected to a tower for ~20 minutes or so.

jpuetthoff, 2017-07-23:
I flew to Oshkosh from KY Saturday. Whenever the tower went from 1 to 0, I got the message: ADS-B disconnected. However, towers came right back to 2-5 and I NEVER lost traffic this time around. Device never disconnected, never lost stratux wifi. The disconnects lasted mere seconds. Using most recent release. The previous release I used I lost traffic, I thought.

cyoung, 2017-07-24:
Is anyone experiencing disconnects outside of ForeFlight?

I did have this happen with Scout. Have not tried Stratux with FF 9.2+ too much.

v1.2r2 has been out for a while and has been pretty stable.

jpuetthoff, 2017-07-24:
I was almost assuming that i was too low to receive towers and hence the ads-b disconnected message. It never seemed to disconnect, but rather not have tower information.

Walkinghispath, 2017-07-24:
I had several yesterday on the way to Oshkosh. Running 1.3b3.

-Jesse

cyoung, 2017-07-24:
@Walkinghispath - ForeFlight 9.2+?

rteabeault, 2017-07-25:
We are seeing this same issue after changing out our high gain SDRs for a pingEFB.
Stratux: 1.4r1
Foreflight: 9.2.1

pmaag, 2017-07-25:
Heard back from Foreflight, the "ADS-B Disconnected" alert occurs when speed > 40kts and a loss of ADS-B data for a short period of time.

I don't believe that there is a loss in ADS-B data and I am going to take a tcpdump on the Stratux to try and show that it isn't occurring (or maybe it is).

Will keep folks posted.

eadasiak, 2017-07-31:
Like @rteabeault - I had this happen with the pingEFB, and according to the stratux logs it was a panic `slice bounds out of range` during `_/root/stratux/uatparse.(*UATMsg).DecodeUplink(0x10955400, 0x0, 0x0)`.  uAvionix support sent me a new unit, and it's been working well both under 1.2r2 and 1.4r1.

Hope I'm not conflating issues between pingEFB vs. sdr's and the disconnects, but wanted to put this out there.

pmaag, 2017-07-31:
Hey @eadasiak , which logfile did you get that panic error in?  I have not had a chance yet to take a packet capture of the issue with Foreflight, but maybe there is a log file I am not looking at.

eadasiak, 2017-08-01:
@pmaag - check `/var/log/daemon.log` -- that was where I was ultimately able to track it down.  Hope this helps!

roperf15, 2017-08-01:
I have been having the same problem since back in the 1.1r1 days. Deleting the log file from the developer page seems to fix it. I agree that it seems to be worse when there is a lot of traffic (i.e. Messages) being processed by the stratux. My non-programmer experience is that it seems to be tied to an excessive amount of 'message queue overflow' messages my stratux starts to generate about 1.0 hrs into flight. My system is a dual blue sdr stratux plus gps currently running 1.3r3 displayed on ipad3 with FF newest.

pmaag, 2017-08-01:
@eadasiak - My /var/log/daemon.log is clean from those errors, looks like our problems might be different.  Thanks for the lead though!

@roperf15 - I do see a ton of "Message Queue Overflow" messages in my logs, both during when the issue occurs and at other times when Foreflight displays the traffic just fine.  The strange thing is that I am seeing Message Queue Overflow messages for other devices that I have had previously connected, but are not connected at the time of the error.  I wonder if stratux is not properly handling clients in the DHCP lease file?  Looking at the lease file it looks like clients will persist in that file after re-boots, 

Looking at the getDHCPLeases() method in network.go, it just reads in the file, which would cause old clients to potentially receive messages that are not even on the network.  I need to spend some more time with the code, but my concern here is that we are sending data out on wlan0 to client IPs not connected to Stratux at all.

Regardless, I am going to try and grab a packet capture off the Pi on wlan0 tomorrow when/if the issue occurs.  That will at least allow us to determine if this issue is a Foreflight problem or a Stratux problem.

cyoung, 2017-08-18:
Consolidate to #631.

closed 2017-08-18 by cyoung

brnnrubr, 2017-10-17:
First time poster....found this page through a Google search trying to find an answer to this problem. 

I’ve been experiencing the same frustrating ADS-B disconnect issue. I wonder if FF 9.4.2 that just released will help.

cyoung, 2017-10-17:
Hi @brnnrubr, thanks for posting. DIdn't even know there was a 9.4.2. Will give that a try. Click on #631 to continue the conversation there.

brnnrubr, 2017-10-17:
It literally just popped up this evening. Hope it clears up the issues.

ccostes, 2017-10-25:
Flew the other day with 9.4.2 and still had issues.

cyoung, 2017-10-25:
Upgrade to 9.4.3. That should fix it.

ccostes, 2017-10-25:
Oh wow, just came out 6 hours ago. Will update and see if it works better on the next flight.

Thanks for the heads up!