cyoung/stratux #643: Weather not displayed

mreddy0817, 2017-07-20:
1. Stratux version:1.2r
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [y] yes dual 160
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [x] no

    power source:battery brick

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)foreflight latest

    EFB platform: (e.g., iOS 9.2) foreflight 9.2.1

    EFB hardware: (e.g., iPad Mini 2) iPad 2 air light 

4. Description of your issue: cannot see weather!!!

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2017-07-22:
If you are on the ground that is typical. The ADS-B ground towers are setup to maximise coverage while airborne.

mreddy0817, 2017-07-22:
Helno, thanks, but I was up today with 4 towers registered, I could not get any weather.  also, my foreflight app was limited only to "Radar", Radar composite Low, and satellite, and color satellite were not lit up (available as choices).

Also, I could not get targets till I turned off "Hide distant targets".

any ideas?

Helno, 2017-07-22:
Sounds like it is working. 

The reason you see the different options is because ADS-B has a limited data set. No satelite and only nexrad radar so no lowest tilt. If you click on nearby airports you will see that it has the latest metars for them. 

As far as traffic goes hiding distant traffic limits you to seeing very close traffic.

mreddy0817, 2017-07-23:
Helno,
 
I am expecting to see the red, yellow and green ( for weather rain clouds) formations the way foreflight shows through internet cell connection.  I do not have I built GPS or cell coverage for my iPad.  I am unable to get this data displayed and the 978 UAT shows no data received.
 
what is the problem?
 
 
 
 
mahesh

cyoung, 2017-08-04:
Could you take a screenshot of http://192.168.10.1/ after flying or while flying?

mreddy0817, 2017-08-04:
Okay, give me a couple of days.

mreddy0817, 2017-08-13:
Chris,
 
see attacted screen shots
 
mahesh

cyoung, 2017-08-13:
@mreddy0817 - they didn't come through, would you mind trying again please?

mreddy0817, 2017-08-13:
Let me know if you get these, they are png files

mreddy0817, 2017-08-14:
Chris,
 
did you get me screen shots.  Do you think that it is a bad 978 chip?
 
Mahesh

cyoung, 2017-08-18:
@mreddy0817 - the screenshots never came through. If you're showing "4 towers" in ForeFlight then you are receiving 978 data. I think it's an issue that needs to be worked out with the layers. Composite radar is the only layer that will show ADS-B data. [layers dropdown](https://foreflight.files.wordpress.com/2017/01/tops-selections1.jpg).

It will sometimes take 10-15 minutes for the radar data to populate as it is not transmitted often. Check it again after receiving for about that amount of time.

closed 2017-08-18 by cyoung