cyoung/stratux #644: Stratux interface software

JimR2017, 2017-07-20:
1. Stratux version:Home built with RPi3
2. Stratux config:

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ x] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
We are developing a new EFB simulation for a research project.  We are looking for more information on accessing data from the Stratux.  Is there a Software Development Kit (SDK) or interface manual that will describe how to structure the interface with the Stratux RPi3?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-07-21:
Hi Jim,

Did you find this link? https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md

If you've already seen it, what further information do you need specifically?

closed 2017-08-04 by cyoung

JimR2017, 2017-08-13:
Thank you for your help.  I think we are making good progress now.
Best
Jim