cyoung/stratux #646: no Weather

z00mie, 2017-07-21:
1. Stratux version:V1.4r1
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type:USB u-Blox 7

    AHRS
      - [ x] yes
      - [ ] no

    power source:
2240mah lithium ion EC Technology external battery pack
    usb cable: with battery

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Fly_Q  version 2.4.5 (10)

    EFB platform: (e.g., iOS 9.2)IOS 10.3.2

    EFB hardware: (e.g., iPad Mini 2) Ipad Pro 9.7

4. Description of your issue: On ground following unit assembly no UAT messages, no weather, recent clients 0, sdr devices 2. Is this a normal or should I receive weather etc on the ground? Oh, also no towers picked up. However the 1090 sdr provides plenty of traffic.

Thanks Z00m

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

peepsnet, 2017-07-21:
No weather if no towers. No towers if not close to an airport... or tower location. I usually need to be within 1 mile of a tower if on the ground.

⁣Sent from BlueMail ​

mreddy0817, 2017-07-22:
I have the same problem, I got no weather while airborne with 4 towers picked up, not sure if it is a setting issue in Stratux or Foreflight?

Austinthemighty, 2017-07-27:
the air ports may not be transmitting the weather go to a large airport near you or one that you know transmits the weather and see if you get any weather data it may take a few minutes

cyoung, 2017-08-18:
See towers here: http://towers.stratux.me

If you're not within ~5 miles or less of the tower on the ground, don't expect to pick it up. It is normal to get no UAT on the ground.

closed 2017-08-18 by cyoung