cyoung/stratux #648: Need flarm method of AvSquirel over cyoung release 1.4r1

ecsaulnier, 2017-07-22:
1. Stratux version: 1.4r1
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type: VK-162

    AHRS
      - [x ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Air Navigation Pro Last Version

    EFB platform: (e.g., iOS 9.2) IOS 10.3

    EFB hardware: (e.g., iPad Mini 2) iPhone 6+

4. Description of your issue:

Hello,
For my EFB App "Air Navigation Pro" i need Flarm module of AvSquirrel.

How install this over Stratux version 1.4r1 or Is-it possible to obtain a release version with the module flarm ?

Thank for your response. 
French User.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2018-01-17:
@ecsaulnier - I'm not sure what the modifications were to the code to make it produce FLARM output. As far as I can tell, AvSquirrel has abandoned the FLARM code - so I don't think we can help much at this point.

closed 2018-01-17 by cyoung

zittix, 2018-03-12:
FYI you can follow those instructions to get 1.4 version to work with air nav pro: https://github.com/0sander/stratux_air_nav_pro

zittix, 2018-03-12:
@cyoung would you accept a patch to set those settings, mentioned in the above link, from the web ui?

cyoung, 2018-03-12:
@zittix - did you attempt to just change the port to 43211 instead of changing the IP also? If it's just a port change, that would be much easier to implement.

zittix, 2018-03-12:
I did the change for both IP and port. I guess making a GUI to configure both would make sense.

peterclubroad, 2018-03-12:
Great news that this is going on. @zittix, has this also been tried with the airconnect device? In other words does anyone know the required setting to be able to use Stratux as a source in setups where an airconnect device is normally needed?

zittix, 2018-03-13:
No I haven’t tried with air connect as I don’t own any. As far as I know air connect expect a Flarm protocol, doesn’t it?

If yes, I am afraid it won’t work as is as stratux only supports GDL90.

Mathieu

ecsaulnier, 2018-03-14:
Hello,
 
Thank you for all. That goes Air Navigation Pro very of course.

It remains for me right to be seen how to filter the planes which are much higher than the miens because making only VFR with 2500 feet, I do not need to see the planes being with the FL200 level.

Cordially.
Eric