cyoung/stratux #649: GPS not working

garyemiller, 2017-07-25:
1. Stratux version: 1.4r1

2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ X] yes
      - [ ] no
      type: uBlox 6 USB

    AHRS
      - [ X] yes
      - [ ] no

    power source:
        USB, with volt/ammeter so known good.

    usb cable: 20AWG

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
         N/A
    EFB platform: (e.g., iOS 9.2)
         N/A
    EFB hardware: (e.g., iPad Mini 2)
         N/A
4. Description of your issue:
        cgps missing.  I cant find any standard Raspbian GPS tools...

        no fix.

 I by uBlox6/78 by the box full, they have always worked for me.  It works in the same location on other
RasPi running plain Raspbian.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

2017/07/25 03:51:50 reading standard input: bufio.Scanner: token too long
2017/07/25 03:51:52 Using SiRFIV config.
2017/07/25 03:52:02 stats [started: 2 hours ago]
2017/07/25 03:52:02  - Disk bytes used = 1.6 GB (85.8 %), Disk bytes free = 264 MB (14.2 %)
2017/07/25 03:52:02  - CPUTemp=58.00 deg C, MemStats.Alloc=2.9 MB, MemStats.Sys=9.9 MB, totalNetworkMessagesSent=0
2017/07/25 03:52:02  - CPU load 
2017/07/25 03:52:02  - UAT/min 180/201 [maxSS=38.80%], ES/min 1/463, Total traffic targets tracked=23
2017/07/25 03:52:02  - Network data messages sent: 0 total, 0 nonqueueable.  Network data bytes sent: 0 total, 0 nonqueueable.
2017/07/25 03:52:02  - Last GPS fix: 2 hours ago, GPS solution type: 0 using 0 satellites (0/0 seen/tracked), NACp: 0, est accuracy 999999.00 m
2017/07/25 03:52:02  - GPS vertical velocity: 0.00 ft/sec; GPS vertical accuracy: 999999 m
2017/07/25 03:52:02  - Last IMU read: now
2017/07/25 03:52:02  - Last BMP read: now

Uptime is 2 hours, so the last fix 2 hours ago is a lie.

cyoung, 2017-07-26:
USB or hooked to the RPi GPIO?

garyemiller, 2017-07-26:
All tried on USB.  

I plugged in a Holux GR-213 (SiRF III) and it worked fine.

I went back to a Navisys GR-601W (uBlox 6, w/ PPS). and spent more time with it.  

Both those GPS are used by many people with gpsd with no issues.

For some reason the GR-601W likes to revert to 9600, not sure what triggers it.  So that was the first problem.  I installed gpsd and used gpsctl to set the GR-601W to 38400.

But Stratux still not working with it.  So I used gpsctl to set the uBlox to NMEA mode.  Then Stratux
saw the device.

First, I recommend that your Stratux img file ship with gpsd so simple problems like GPS configuration can be dealt with easily.  gpsd is in the Raspbian repo: "apt-get install gpsd"

Second, gpsd auto-configures to pretty much every known GPS type, stop reinventing the wheel and just use gpsd in Stratux.

Until you get better uBlox support I'll have to stick with the Holux.

kdknigga, 2017-07-27:
There's a gpsd development branch.

https://github.com/cyoung/stratux/tree/gpsd

It could probably use some love.

garyemiller, 2017-07-27:
I know gpsd inside and out, but I don't understand go, or the structure of stratux, or its development process.

I'd be eager to help if you could point me at specific problems that I could solve in C or Python.

cyoung, 2017-08-18:
Closing this since it is concerning a collection of unrecommended hardware. Both the Navisys and Holux don't seem to be readily available and cost $45+, and thus likely will never be very widely adopted.

closed 2017-08-18 by cyoung

garyemiller, 2017-08-19:
The Holux is the one the worked.  It is the Navisis that does not.

$45?  I'll sell you all you want for a lot cheaper than that, plus the have native PPS do also good for semi precision time keeping.   Nothing special about the Navisys, it is a plain vanilla uBlox, so representative of the majority of the better/cheaper GPS on the market.