cyoung/stratux #650: TypeError: __init__() got an unexpected keyword argument 'port'

garyemiller, 2017-07-25:
1. Stratux version: 1.4r1

2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ X] yes
      - [ ] no
      type: uBlox6, 7 and 8.  generic USB

    AHRS
      - [ X] yes
      - [ ] no

    power source:
         USB, with volt/ammeter, so I know the power is good.

    usb cable:
         20AWG

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
       N/A
    EFB platform: (e.g., iOS 9.2)
       N/A
    EFB hardware: (e.g., iPad Mini 2)
      N/A

4. Description of your issue:

Brand new install, I get this on startup:

# cat /var/log/stratux-screen.log
started with pid 600
Traceback (most recent call last):
  File "/usr/bin/stratux-screen.py", line 92, in <module>
    daemon_runner.do_action()
  File "/usr/lib/pymodules/python2.7/daemon/runner.py", line 186, in do_action
    func(self)
  File "/usr/lib/pymodules/python2.7/daemon/runner.py", line 131, in _start
    self.app.run()
  File "/usr/bin/stratux-screen.py", line 23, in run
    oled = ssd1306(port=1, address=0x3C)
TypeError: __init__() got an unexpected keyword argument 'port'

cyoung, 2017-07-26:
Looks like another error with that ssd1306 Python package. It won't affect your build in any way unless you have an OLED screen that you want to display data on.

garyemiller, 2017-07-26:
I see pylint has more to say on stratux-screen.py:

```
# pylint stratux-screen.py 
No config file found, using default configuration
************* Module stratux-screen
C: 42, 0: Trailing whitespace (trailing-whitespace)
C: 49, 0: Unnecessary parens after 'if' keyword (superfluous-parens)
C: 58, 0: Line too long (81/80) (line-too-long)
C: 61, 0: Line too long (97/80) (line-too-long)
C: 62, 0: Line too long (146/80) (line-too-long)
C: 65, 0: Line too long (113/80) (line-too-long)
C: 70, 0: Line too long (95/80) (line-too-long)
C: 71, 0: Line too long (146/80) (line-too-long)
C: 74, 0: Line too long (112/80) (line-too-long)
C: 81, 0: Line too long (165/80) (line-too-long)
C: 82, 0: Line too long (84/80) (line-too-long)
C:  1, 0: Invalid module name "stratux-screen" (invalid-name)
C:  1, 0: Missing module docstring (missing-docstring)
C: 14, 0: Missing class docstring (missing-docstring)
C: 14, 0: Old-style class defined. (old-style-class)
C: 21, 4: Missing method docstring (missing-docstring)
R: 21, 4: Too many local variables (24/15) (too-many-locals)
E: 23,15: Unexpected keyword argument 'port' in constructor call (unexpected-keyword-arg)
E: 23,15: Unexpected keyword argument 'address' in constructor call (unexpected-keyword-arg)
C: 30, 8: Invalid variable name "n" (invalid-name)
C: 35,12: Invalid variable name "getStatusHTML" (invalid-name)
C: 36,12: Invalid variable name "getStatusData" (invalid-name)
C: 37,12: Invalid variable name "CPUTemp" (invalid-name)
C: 44,12: Invalid variable name "getTowersHTML" (invalid-name)
C: 45,12: Invalid variable name "getTowersData" (invalid-name)
C: 46,12: Invalid variable name "NumTowers" (invalid-name)
C: 47,16: Invalid variable name "towerLatLng" (invalid-name)
C: 50,20: Invalid variable name "NumTowers" (invalid-name)
C: 77,16: Invalid variable name "t" (invalid-name)
C: 79,20: Invalid variable name "t" (invalid-name)
C: 81,20: Invalid variable name "t" (invalid-name)
C: 83,24: Invalid variable name "t" (invalid-name)
C: 87,12: Invalid variable name "n" (invalid-name)
R: 21, 4: Method could be a function (no-self-use)
R: 21, 4: Too many statements (52/50) (too-many-statements)
R: 14, 0: Too few public methods (1/2) (too-few-public-methods)
C: 90, 0: Invalid constant name "stratuxscreen" (invalid-name)
C: 91, 0: Invalid constant name "daemon_runner" (invalid-name)
W:  3, 0: Unused import sh1106 (unused-import)
W:  6, 0: Unused import ImageDraw (unused-import)

Report
======
72 statements analysed.

Raw metrics
-----------

+----------+-------+------+---------+-----------+
|type      |number |%     |previous |difference |
+==========+=======+======+=========+===========+
|code      |71     |87.65 |NC       |NC         |
+----------+-------+------+---------+-----------+
|docstring |0      |0.00  |NC       |NC         |
+----------+-------+------+---------+-----------+
|comment   |1      |1.23  |NC       |NC         |
+----------+-------+------+---------+-----------+
|empty     |9      |11.11 |NC       |NC         |
+----------+-------+------+---------+-----------+

External dependencies
---------------------
::

[... 61 lines cut ...]
+-----------------------+------------+
|too-few-public-methods |1           |
+-----------------------+------------+
|superfluous-parens     |1           |
+-----------------------+------------+
|old-style-class        |1           |
+-----------------------+------------+
|no-self-use            |1           |
+-----------------------+------------+

Global evaluation
-----------------
Your code has been rated at 3.33/10

Duplication
-----------

+-------------------------+------+---------+-----------+
|                         |now   |previous |difference |
+=========================+======+=========+===========+
|nb duplicated lines      |0     |NC       |NC         |
+-------------------------+------+---------+-----------+
|percent duplicated lines |0.000 |NC       |NC         |
+-------------------------+------+---------+-----------+
```

penniesfromkevin, 2018-01-10:
@garyemiller Did you ever resolve this issue?  I took a look at it and have a new file here: https://github.com/penniesfromkevin/stratux/tree/master/test/screen
Even if it is no longer an issue, if you have a setup using a display can you test the changes?  I would appreciate it greatly!

garyemiller, 2018-01-11:
@penniesfromkevin I'm on the road until Monday.  I'll check then.

penniesfromkevin, 2018-01-12:
Great!  Have a safe trip!  Can I ask how you set up your display?