cyoung/stratux #651: Getting Stratux Information

Sara1371, 2017-07-25:
1. Stratux version: home-built with RPI3
2. Stratux config:

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [x
[image]
[image]

 ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
I want to get information through this link as an example. http://192.168.10.1/getSituation
, but what I get instead of picture 1 is picture 2. What is the reason?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Nokomis449, 2017-07-25:
Don't know if this is helpful, but you can see how the (old?) Python screen script pulls the data by looking at the file https://github.com/cyoung/stratux/blob/master/test/screen/screen.py

Specifically:
            response = urllib2.urlopen('http://localhost/getStatus')
            getStatusHTML = response.read()
            getStatusData = json.loads(getStatusHTML)
            CPUTemp = getStatusData["CPUTemp"]
            uat_current = getStatusData["UAT_messages_last_minute"]
            uat_max = getStatusData["UAT_messages_max"]
            es_current = getStatusData["ES_messages_last_minute"]
            es_max = getStatusData["ES_messages_max"]

kdknigga, 2017-07-25:
It looks like you got what you wanted, just without pretty formatting of
the json.

Sara1371, 2017-07-25:
Ok. Thanks a lot. Here is my issue: I am looking for the information of the aircraft in the picture attached. I mean their lat, long and etc. What I am getting in python while typing these are lat=0, long=0 and.... I could get nothing related to the aircraft seen in the traffic part of Stratux website. Why am I getting lat =0, lng=0?
[image]
Many thanks for the help.

kdknigga, 2017-07-25:
getSituation shows stratux's location information from the GPS attached.
If you want traffic information check ws://192.168.10.1/traffic

Sara1371, 2017-07-26:
Thanks a lot. I want to see what raw data is coming into Stratux or more specifically Raspberry Pi, that then is converted to the format we can see in 192.168.10.1/traffic website. I want to see any information which an ADS_B message contains. (all information as location, address, ICAO, velocity, etc. Is there any way for me to get that?

kdknigga, 2017-07-26:
Well, then you're probably more interested in dump978 and dump1090.  These
are the programs/libraries Stratux uses to tune the radios and decode the
messages.

https://github.com/mutability/dump978
https://github.com/antirez/dump1090

Sara1371, 2017-07-26:
Dear kdknigga,
Thanks a lot for your help. I looked up at those links you gave and it seems they are exactly what I am looking for a long time. I need this data for part of my research at Electrical Engineering department so, I am a beginner in this particular area. Is there a possibility for you to give a more detailed guide about where I should start? Is there any specific program that I should install? As I am not familiar with Stratux related programming, the explanation written is not clear for me.

kdknigga, 2017-07-26:
The dump1090 readme is pretty straightforward.  What specifically are you
having problems with?

Sara1371, 2017-07-26:
Where should I run this program? Does it need special programming tool? I can see nothing written about that.

cyoung, 2017-07-26:
For dump1090 and dump978 you'll have to install the programs and their dependencies.

It really depends what you mean by "raw data". The "raw data" might just be a radio recording. The `/traffic` websocket passes through an update every time it gets data over the air. But it does combine messages like altitude updates with the last known (lat,lng).

https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md

See the section on part at the bottom, `ws://192.168.10.1/traffic`.

Sara1371, 2017-07-28:
Hello again. I am now on Linux trying to go through the dump-1090 instructions. I downloaded the related zip file and tried to go through the instructions given. When I typed "make" it said you do not have librtlsdr.pc on your PC. I am attaching the screen shot here. Is there any other extra file that I should have beside dump1090-master file?
[image]
Many thanks for the help.

Bearcode, 2017-07-28:
@Sara1371 This looks like a fix for your specific error: https://github.com/antirez/dump1090/issues/108

This guide will get you the rest of the way https://ferrancasanovas.wordpress.com/2013/09/26/dump1090-installation/

Also checkout [Virtual Radar Server](http://www.virtualradarserver.co.uk/) for real-time visualization of this data. If you're not going to be flying with your Stratux device a then something like this plus dump1090 might be an easier setup to work with. Otherwise, if you stick with the Stratux device you can enable "Replay Logs" and just use SQL to get the data out of `/var/log/stratux.sqlite` with a select statement like this: 

`select traffic.Reg, traffic.Tail, traffic.Icao_addr, traffic.TargetType, traffic.Lng, traffic.Lat, traffic.Alt, timestamp.GPSClock_value FROM traffic INNER JOIN timestamp ON traffic.timestamp_id=timestamp.id;`

Sara1371, 2017-08-02:
@Bearcode 
Thank you very much. This was exactly what I was looking for. I have another question I wonder if you can help me. I want to store the data which I am getting through ./dump1090 --raw. Is there any possibility for doing that?

cyoung, 2017-08-04:
The sqlite logs that @Bearcode references should have that data.

Sara1371, 2017-08-04:
@cyoung 
Thanks for your help. Actually I am seeking a way to be able to store or save the raw data when displaying on Rpi's terminal and decode the file with a program written by myself. Is this possible to save the file? As far as I understood from @Bearcode 's comment, sqlite is for when we are using Stratux, but know I am using ./dump1090 --raw to get the data.
Thanks a lot for your time.

cyoung, 2017-08-18:
@Sara1371 - looks like you've got information needed for your project, re-open if you need anything further.

closed 2017-08-18 by cyoung