cyoung/stratux #652: Any interest in me sending in a pull request that adds BeagleBone Blue support?

geeksville, 2017-07-28:
Hi,

I wasn't sure if there was a better mailing list to ask this on, but before I go cleaning up a PR I thought I'd check if ya'll have any interest in adding [BeagleBone Blue](https://beagleboard.org/blue) support?  I made a small number of minor changes so that Stratux could work on BB Blue or the standard rPi.  BB Blue is nice because it includes a 9250 IMU and a BMP280 baro on the board.  So totally self contained, just add GPS and SDR and you can go.

cyoung, 2017-07-28:
@geeksville - thanks for asking. That belongs more in github.com/jpoirier/stratux-setup.

closed 2017-07-28 by cyoung

geeksville, 2017-07-28:
okay - Though I did need to change some of the go code (because devices on different i2c busses).  Still the right place to ask?

cyoung, 2017-07-28:
I think it should compile and run without any modifications to the code. We use the embd go i2c library which handles the differences between Raspberry Pi and Beaglebone.

geeksville, 2017-07-28:
Almost but not quite ;-).  Expressions similar to this:

https://github.com/cyoung/stratux/blob/master/main/sensors.go#L35

Cause problems because though these sensors are on i2c bus #1 on a rPi, they are on bus #2 of a beaglebone.  The fix I made is fairly minimal, which is to attempt finding the device on bus one, but if not found probe on bus 2. ;-)

cyoung, 2017-07-29:
Interesting, are there any other changes needed?

geeksville, 2017-07-29:
No - not for the go code, the scripty goo you alluded too in github.com/jpoirier/stratux-setup of course will need a small number of changes.  So do you want me to make up a PR?  It should run fine on rPi or bbone with this change.