cyoung/stratux #654: Constant SD card corruption after proper shutdown

ismarhernandez, 2017-08-02:
1. Stratux version:     stratux-v1.3b1-a6f9aaf9f5-dev.img
2. Stratux config:      Raspberry Pi3  - NooElec Dual-Band NESDR Nano 2 ADS-B (978MHz UAT & 1090MHz 1090ES) - Stratux AHRS sensor - External USB GPS antenna WAAS capable - Summitlink tinted case with fan - I flash the stratux software to an SD card using my MAC. 

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [X] yes
      - [ ] no

    power source: USB cable to car carger - I will be ordering a 22000 mAh battery soon

    usb cable: regular USB cable to stratux 

3. EFB app and version: FLTPLAN Go

    EFB platform: iOS 10.3.3 (iPad Pro 9.7 Tablet and iPhone)

    EFB hardware: (iPad Pro 9.7" and iPhone 7plus )

4. Description of your issue: Hey guys, since I knew about Stratux I knew it was a great system, so I went ahead and I built my own. I have great reception, GPS, AHRS and traffic information. But my Stratux box is not very reliable since the SD card keeps getting corrupted constantly. Let's say I use the Stratux today, and after the Stratux starts I see the fan start, the green light blinking rapidly (about 5 times per second) and no red light shown. However, the Stratux works flawlessly... beautifully!!! It gets traffic better than my friend's Stratus, great GPS, and AHRS information. I then conduct a proper shutdown through the 192.168.10.1 and clicking the "SHUT DOWN" button. Immediately, the red and green light comes on steady, and my fan stops working; perfect, I think I turned it off correctly. I give it 10 seconds after this and then disconnect the USB cable at the Stratux -no problems so far. 
The problem comes after the second time I use the Stratux. I do the same thing to start it like the last time by connecting the cable to the electricity, then to the Stratux. However, I do not find the Stratux wifi signal and I am not able to connect to it. The green light stays blinking the same way as if it was working, but I do not see the wifi signal. I have attempted this as a controlled experiment using wall plug adapter(iPhone charger)-to-USB cable-to-Stratux, cigarette lighter adapter-to-cable-to-Stratux, with the same results. I have attempted to find the signal with a wide array of things (Mac, iPad, iPhone, other computers, and phones), to no avail. After this, there is no way I can get the Stratux to work unless I re-flash the SD card. When I reflash the card and use the Stratux, it works as if it was brand new again. 
I read a post that explains the SD card may be getting corrupted and a brief explanation on how to make the SD card "read-only". However, I don't seem to be able to make it work. I don't think to be following the instructions correctly and messing up somewhere. I am able to use a computer and follow steps to the exact letter (including using terminal and prompt), but I am not a programmer and do not have a lot of experience working with this. I can follow the instructions, but they have to be broken down as if I was an 8 year old child. 
Could someone please extend me a hand with this problem? 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

peepsnet, 2017-08-02:
SD card corruption Is common with underpowered supplies. If your power supply is not providing over to amps you're likely to have that corruption quite constantly. I lost about four cards to plugging it into a standard car charger or USB port in my car.

 You definitely want to get a 223 amp charger just to make sure

ismarhernandez, 2017-08-02:
I understand where you're coming from, but as I mentioned in the post, I tried this using several power sources, including a wall plug adapter giving me the correct voltage and amps.

Nokomis449, 2017-08-02:
Per this, a green blinking light 10 times per second indicates a system error.  I suggest you re-image your SD card and see if the problem goes away.  It may be as simple as your card being full, especially if you have logging turned on, especially if you didn't expand your card.
https://github.com/cyoung/stratux/commit/176cf42f2e8a03852456782ac4e3903d703fca30

ismarhernandez, 2017-08-02:
I have to constantly flash the card to get it working again. logging is not turned on... I dont understand the part of expanding the card... i have a 16gb card on, there's plenty of space.

Nokomis449, 2017-08-02:
You have a 16G card, but the image you're putting on it is about 2G.  So unless you expand the partition, your system only "sees" 2G.  If you know how to SSH, Google how to expand a Pi card and you'll find an easy tutorial.

ismarhernandez, 2017-08-02:
done, but it just did it to me on wall power.

ismarhernandez, 2017-08-02:
on the 192.168.10.1 says i still have space on my sd card. I just dont know why is the SD card constantly getting corrupted. it doesnt matter how many methods i use to read only that card, i just dont seem to either understand, or they just fail

Nokomis449, 2017-08-02:
Don't assume that it's adequate just because it's a wall charger.  Chargers are meant to CHARGE things, not POWER things.  You can charge your car battery with a 1/2amp charger just as well as you can charge it with a 10amp charger; it'll just take a lot longer.  A lot of 5v phone chargers, wall plugs included, are rated for 1/2amp or so, and is fine for charging a phone overnight.  But a Pi in the Stratux configuration needs over an amp of power at times.  If the RED light isn't on, then the Pi isn't getting enough power.  And just as important as the power source is the cable.  You can try to jumpstart a car using 8ga cables and it might work and it might not, but if you put some thicker cables on there, it'll fire right up. Same with USB cords.  They come in different sizes, and a long cable or a smaller gauge cable is not going to be able to supply the amps that the Stratux needs, regardless of power source.  Your initial post says you don't have a RED light.  That's where I'd start.  Anker is a good brand for power AND cables, and there are others.  If you can SSH into it, check the log files in /var/log and see if there's a clue there.  Your problem seems to be unique to you, so start with the red light and go from there.  Also, the green light should be steady, but since it's fast blinking it is screaming "I'M SICK!".  The fan is a seperate program from Stratux, so just because the fan starts up at boot doesn't imply that Stratux is starting up.  Good luck!

Nokomis449, 2017-08-02:
Well the red light is a Pi thing, not a Stratux thing.  The red light shuts off if the Pi voltage drops below (I think) 4.8 volts.  So if you don't have a red light and don't have low voltage, then your Pi has a problem.

ismarhernandez, 2017-08-02:
thank you so much Nokomis449... the pi is working fine. I may have this issue after one use or 2 uses, but I get the issue where I have to constantly reflash the card... at this point the PI issue is discarded. I just needed some help with making my SD card READ ONLY.... does anyone have any ideas in baby steps? has anyone tried this before?

rhole, 2017-08-02:
Could it be a defective SD card?  You might try a different card.

Rick & Sonia Hole
Email: r.hole@ieee.org<mailto:r.hole@ieee.org>

ismarhernandez, 2017-08-02:
well, I thought of that also... I went ahead a purchased two more cards. Another 16gb card and a 32gb card. it still does the same thing. The pi doesn't have any issues and it's not underpowered. Thank you so much for your help rhole... for now, the last thing I want to try is to make my SD card read only. 
If anyone know how to make an SD "read-only" and you don't mind sharing your knowledge, please let me know. I'm not very computer savvy, but I can follow instructions to the letter if you provide me with those.

rhole, 2017-08-02:
Have you proved it has sufficient power?  Others have had similar problems solved by power supply or power cord upgrades. What counts is power at the RPi. The SDRs draw considerable power. If you could measure the 5 volt at RPi to be sure it has proper voltage you will settle the issue.
Canakit offers a wall wart supply at 5.2v because many RPi users have seen power related issues.
I have a not-Stratux application and set the voltage at 5.1 volts as a preventative measure.
The Stratux community wants you to succeed!
By the way, if you connect an HDMI monitor, USB keyboard and mouse you will boot up with a graphic interface and can query the two volumes on the SD card to see what free space is there.
When you load the SD card from an image the SD becomes the size of the image (about 2gb) regardless of the physical size of the card. You must expand the file system to make the space above 2gb usable.
Making a read only RPi is a complex task. I am not a Stratux developer and do not know if Stratux must have read/write access to the SD file system to run. Perhaps developers can answer that question.

Rick Hole

cyoung, 2017-08-02:
@ismarhernandez - you experience this error with a brand new SD card? An SD card that is physically damaged can cause erratic behavior that is "fixed" temporarily by re-imaging.

ismarhernandez, 2017-08-03:
I have purchased 3 different cards. two 16gb cards and one 32gb card. these behave in the same way. I had a friend of mine check these cards, and he said there was no problem. I then installed the 32gb card in a picture portrait display and filled all 32gb with images with pictures. about a month 15 days later I tried accessing the pictures and there was not an issue. The issue comes when I flash the Stratux software to the SD card and begin using it.  After the first of second use, the stratux does not restart at all. I can leave it connected for an hour, connect it to a viable power supply, switch cables, dance around it, but it will not start. The only way to re-start it is by physically taking out the SD card, reflashing the SD with the stratux 1.3 software, and inserting it again. After this, I can connect the stratux to any power source, even if it's a 1 amp (I don't do this, but I did it once) it WILL work. Once it starts working, it will not quit. I have left the unit working for 24h without an issue.... I then turned it off and let it rest for about 5 hours. then I turned it on for an hour and it worked just fine, and on a different power source. That day I went to fly and it did not work (wi-fi) not showing up. For some reason, stratux is corrupting the SD card... I heard the best solution is to configure the SD card to be read only, that way, the SD card cannot be overwritten by the stratux and the files cannot be corrupted.

cyoung, 2017-08-03:
From what you've written it's hard to tell exactly what is going on. Can you confirm that you have re-imaged three different microSD cards with stratux, and each of these three microSD cards have individually become corrupted?

ismarhernandez, 2017-08-03:
yes, three different SD cards... one 32gb card, and two 16gb cards... I have had to reimaged each card, probably about 6-8 times (each card) within the last month.

jzeevi, 2017-08-03:
Have you actually looked at the cards to make sure they really are corrupted and there’s not some other issue? (marginal HW, etc)

ismarhernandez, 2017-08-03:
how do you know if they are corrupted? Im having this idea of being corrupted because re-flashing fixes the issue AND I have checked the stratux and it has appropriate power, AND I have used three good cards. is there a way to find out exactly? I have a MAC

jzeevi, 2017-08-03:
You should be able to mount them on the MAC (similar to what you do when flashing it) and see it you can do an ls to see the files (there are two sections, as I recall – a boot section and the main file system).

ismarhernandez, 2017-08-03:
gotcha... I can only see one section. for the most part I dont have any folders visible. however, i can see these folders if i select to view hidden folders. after doing so, I am not able to access the two other hidden folders. :/

jzeevi, 2017-08-03:
Er, what do you mean “after doing so, …”

ismarhernandez, 2017-08-03:
after displaying the hidden folders

jzeevi, 2017-08-03:
You’re saying, then, that you can only see the folders one time only. You cannot see them again before you use the card?

ismarhernandez, 2017-08-03:
I can see them before and after using the card, but they appear as hidden folders (ghost looking), and I am unable to access them. However, on the SD card, as soon as I access it I see other files, .sh files, .exe files and other things.

jzeevi, 2017-08-03:
Sounds like that mount point isn’t being affected, then. Those hidden files are normal as part of the system.

 

Maybe someone can offer up a method to mount the secondary file system, or you can view it on the rPi.

ismarhernandez, 2017-08-03:
what do you mean about secondary file system setup -sorry, but I'm not a programmer in any shape or form. However, I can follow instructions to the letter if you guys tell me how.

ismarhernandez, 2017-08-03:
i tried following this: 
Here is a way to make the Stratux Read Only from a Windows PC:

#!/bin/bash

##
## Pi Update / Make-Read-Only Script.
## 
## Download the script files from: https://github.com/cyoung/stratux/issues/78
## When unpacked, you will have a folder named "ropi"
## 
## In the folder downloaded from above, save this script starting from the #!/bin/bash as a text file and name the
## file "setup.sh" replacing the "setup.sh" in the "ropi" directory
## 
## With Filezilla copy "ropi" directory to the pi to /home/pi/ 
## 
## Connect pi to Ethernet for Internet access to install packages in script
##
## SSH to the pi and within the /home/pi/ropi directory and run: sudo bash setup.sh 
## 
## After installation: sudo reboot or sudo halt
##

# Replace the default DHCP server startup file
scp /home/pi/ropi/isc-dhcp-server /etc/init.d/isc-dhcp-server

# Remove the checkroot-bootclean.sh script
update-rc.d -f checkroot-bootclean.sh remove
rm -f /etc/init.d/checkroot-bootclean.sh

# Add the ro / rw commands
scp /home/pi/ropi/ro /usr/bin/ro
scp /home/pi/ropi/rw /usr/bin/rw

# Remove a bunch of stuff we don't need 
apt-get -y remove --purge triggerhappy cron logrotate dphys-swapfile fake-hwclock
apt-get -y autoremove --purge

# Install the busybox version of syslogd
apt-get -y install busybox-syslogd
dpkg --purge rsyslog

# Delete a bunch of stuff to make clean mount points
rm -rf /tmp/*
rm -rf /var/tmp/*
rm -rf /var/log/*
rm -rf /var/lib/dhcp/*

# Replace the /boot/cmdline.txt file with the RO, fastboot version
scp /home/pi/ropi/cmdline.txt /boot/cmdline.txt

# Replace the fstab with our RO version
scp /home/pi/ropi/fstab /etc/fstab
/////////
but I noticed that it did not make the file read-only....

cyoung, 2017-08-03:
Try with the latest image: https://github.com/cyoung/stratux/releases/download/v1.4r1/stratux-v1.4r1-f6bad36cdc.img.zip.

You definitely shouldn't be using the dev image.

ismarhernandez, 2017-08-03:
ok.... Will try with the latest image... Downloading it right now. I will put it to the test.

jzeevi, 2017-08-03:
Also, to test if Stratux is corrupting the image, you can connect to Stratux via the web page (192.168.10.1) go to settings, and do a shutdown before yanking power.

ismarhernandez, 2017-08-03:
yes, that's how I usually shut down the power. first I go to the 192.168.10.1 page, then go to settings, then i select the shutdown button and confirm I want to shut down. I then give it a good 10 seconds after confirming shutdown and then disconnect it.

ismarhernandez, 2017-08-03:
cyoung, what does the "dev" image means? was that one not good?

cyoung, 2017-08-18:
Did the non-dev image work, then? @ismarhernandez

ismarhernandez, 2017-08-18:
@cyoung 

Hello! At this time I have not had any issues. I wanted to test it for this month and see if the problem has been fully fixed. I noticed the red light and the green light are continuously on after I switched the version. The power supply cable, the power supply, and the SD card are the same as before. The only thing I changed was the version. I am starting to think the problem was with the "dev" image, and not with the power supply or SD card.

cyoung, 2017-08-18:
I think that was the case. Re-open if you have further issues. Thanks for responding.

closed 2017-08-18 by cyoung