cyoung/stratux #655: ADS-B Target relative altitudes use pressure altitude, not GPS Altitude

ThatOneRoadie, 2017-08-05:
1. Stratux version: 1.4r1
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: RY836ai GPIO

    AHRS
      - [x] yes
      - [ ] no

    power source: 22000 mAh Battery

    usb cable:

3. EFB app and version: ForeFlight 9.2.1

    EFB platform: iOS 10.3.2

    EFB hardware: iPad Mini 4
4. Description of your issue:
Targets are displayed on the map with relative altitudes in +FL format (e.g. for ownship at FL200 and target at FL320, Target is shown with +120 on arrow). Relative altitudes for these target values are calculated using barometric pressure altitude instead of GPS altitude, leading to targets like http://i.imgur.com/0bUu5ey.png where ownship is at FL410, target is FL360, but target is shown on map as +247

Solution: if GPS source <100m accuracy is available, calculate relative targets via GPS, and failover to pressure if no GPS is available.

cyoung, 2017-08-05:
ForeFlight actually calculates the relative altitudes. Looks like in 9.2 they switched to calculating relative altitudes based on pressure altitude, if available. I tend to think that this should be fixed on the EFB side. Thoughts @bsdcat?
See also #348.

Ergonomicmike, 2017-08-05:
Am thinking that you can't use a Stratux's ownship PA in a pressurized cabin. The internal BMP will think that you're flying at your cabin pressure, instead of your FL altitude.

cyoung, 2017-08-05:
Yes but the EFB should actually make that decision. We send both GPS and pressure altitude to the EFB. There are some EFBs that have depressurization alerts, for example.

bsdcat, 2017-08-05:
To clarify I believe Stratux sends "both GPS and pressure altitude" for ownship, via message `0x0A` for pressure altitude and `0x0B` for GPS altitude. For other traffic, only pressure altitude is reported, via message `0x14`. I think the issue, then, is that ForeFlight is trying to compare like to like - ownship pressure altitude with traffic pressure altitude. I seem to recall that this scenario with a pressurized cabin is something they are aware of, but contacting their support is probably the best way to report this and see it get fixed.

I myself actually just moved on from ForeFlight, so I can't be of any more help I'm afraid.

cyoung, 2017-08-06:
Sent in the bug report and they responded "thank you for the PIREP and your thoughts". I guess we'll see.

sicktim, 2017-08-27:
I agree with this. The issue is that pilots fly whatever pressure alt is showing on their altimeter. This is also reported to ads-b. The issue isn't solely cabin pressurization, but even if using ambient air pressure, there are errors between pressure altitude and gps altitude caused by temperature and other sources. 

My idea is for stratus to identify your aircraft using mode s and use that reported altitude as your altitude for your efb. Then you won't have alt discrepancies.

GitUser275, 2017-11-15:
"The issue is that pilots fly whatever pressure alt is showing on their altimeter. This is also reported to ads-b. "

Sorry but his is not accurate, or is at least incomplete.   Pressure altitude is only shown on an aircraft altimeter when the altimeter baro correction is set to standard pressure.   In USA, pilots will only have the altimeter correction set to standard pressure when flying above 18,000 feet (the transition altitude).  For flight below the transition altitude the altimeter is set to a local correction.

This means that, below the transition altitude, there is usually a difference between pressure altitude and the altitude being indicated on the altimeter.  The pilot uses the indicated altitude, not pressure altitude, when flying below the transition altitude.

ADS-B out includes pressure altitude.  This parameter is derived from pressure measured at one or more static ports that are external to the aircraft and not influenced by changes in cabin pressure.  The same value of pressure altitude is included in the mode C or mode S transponder output.  Air traffic control applies the local baro correction to this received pressure altitude so they know whether an aircraft is flying at the assigned altitude.

A pressure sensor in the cabin of an unpressurized aircraft can give a reasonable estimate of the aircraft pressure altitude -  but only if there is provision for calibrating it so it provides the same altitude reading  as the aircraft altimeter when that altimeter is set to standard pressure.   Even then there will be unknown errors in flight depending on aircraft speed and what air vents are open.

sicktim, 2017-11-20:
Small details. But the big picture still stands. My EFB compares everyone else's ADS-B altitude (pressure related) to my GPS altitude (triangulated altitude) and shows errors from hundreds to thousands of feet.

GitUser275, 2017-11-20:
"and shows errors from hundreds to thousands of feet"

 That shouldn't be a surprise.  ADS-B out traffic reports are referenced to pressure altitude.  Any CDTI should use pressure altitude to compute the relative altitudes of these targets.   If the CDTI has no pressure altitude source, and the CDTI designer chooses to use GPS altitude as a reversion, then there must be a means for the user to apply an offset to the GPS altitude before it is used for target relative altitude display.

sicktim, 2017-11-21:
It is in Foreflight. It would be a great feature for Foreflight or Stratux to use my aircraft's altitude reported through ADS-B and display that on the EFB as the primary altitude source. Then useful altitude comparisons can be made between aircraft. What does CDTI stand for?

GitUser275, 2017-11-21:
CDTI - Cockpit Display of Traffic Information.  The term was commonly used when the ADS-B concept was first introduced but seems to be less commonly used now.

cyoung, 2018-01-17:
Anyone able to test this with a recent ForeFlight version to see if they fixed it?

JohnOCFII, 2018-01-17:
> Anyone able to test this with a recent ForeFlight version to see if they fixed it?

Weather permitting - I can test it this weekend.  Did you spin up a new Dangerzone version too?

cyoung, 2018-01-17:
No new dangerzone now. I'll try to bring the latest commits into the dangerzone branch in the next couple days.

riccrouch, 2018-04-03:
I am having a related similar problem. Because I fly in a pressurized cabin, I am shown traffic relative to my cabin altitude, not my actual altitude. I contacted Foreflight and was told to go to Devices->Stratus Stratus and uncheck "cabin is pressurized." I was also told that if that option didn't appear (it didn't) to update my firmware.

I am new to Stratux, so I don't know how to check the firmware version number to see if I have the latest, so I am going to download the current version and see if it has that option.