cyoung/stratux #656: Set AHRS Sensor Orientation: Failed!

boa888, 2017-08-10:
1. Stratux version:
2. Stratux config:

    SDR
      - [x ] single
      - [ ] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [x] yes
      - [ ] no

    power source:

    usb cable:

3. Table top test

   
4. Description of your issue:

I have the RY836AI in a small case on my desk. I am getting GPS and 1090ES data. In the plane I would mount the module vertically (on edge) with the EMI shield pointing toward the propeller. So I start the Orientation process vertically as it would be in the plane then move it horizontally and press the "Set Forward Direction" . I get the "Set AHRS Sensor Orientation: Failed!" BUT just after I tap "Set Forward Direction". For a nano I see an error message in the lower right corner that I think says DONE but it then goes to the Failed message.  
Sometimes when I lay it down then press the "Set Forward Direction" it just hangs there. 
This happens on the latest release and 7c00

Any help would be appreciated
[image]

able "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

westphae, 2017-08-12:
Thanks for the excellent post with images and stratux.log.  (FWIW you might get a faster response posting to https://stratuxadsb.slack.com/#ahrs).

From the pictures, I believe you are doing the calibration procedure correctly.

From the stratux.log, it appears that your RY836AI isn't connected properly.  There are multiple messages "AHRS Error: Couldn't initialize an IMU" every time it tries to connect.  (It also doesn't connect to the baro sensor on the 836, the BMP280).  Judging from the picture, are you only connecting the RY836AI via USB?  If so, that only connects the GPS.  The IMU and BMP sensor chips have to be connected by wires to the correct pins on the Raspberry Pi headers.  There's some help under the "Requirements" section at https://github.com/cyoung/stratux/wiki/All-About-AHRS

Your logs also show that it isn't finding a /etc/stratux.conf file, which is where your user settings are stored.  This will prevent the orientation (or any other settings) from being saved, which will be a real hassle.  You may have a bad sdcard if you didn't somehow do this intentionally.

It's been on my to-do list to disable the AHRS settings if the chip isn't present or enabled.

I'll give you a few days to reply and verify that the sensors are indeed not connected and then will close this.

boa888, 2017-08-14:
I never saw the "AHRS Error: Couldn't initialize an IMU" message but I did connect the RY836AI directly to the GPIO and it works as expected. However I did use SDA1 and SDL1 to pins 03 & 05. Should I connect SDA2 and SCL2 to pins 03 & 05 or is that instruction only if I was adding a separate board.?

I really appreciate your help very much. I will see this information is posted to Reddit\Stratux.

westphae, 2017-08-15:
Ah, I saw your message about the lag on Reddit and replied there.  Yes, that is way too long.  The response should be nearly instantaneous (10-20 milliseconds lag).  I've typically seen this when my device's browser is lagging a lot.  Try connecting with a laptop or restarting your phone/tablet and then connecting to see if it's any faster.  That large lag could also just be it totally ignoring your sensors if you don't have SDA1/SCL1 connected, and just glitching a bit when it gets some weak GPS signal that makes it think it's moving (there's a separate pseudo-AHRS algorithm that tries to guess attitude based on GPS data only).

The "AHRS Error: Couldn't initialize an IMU" messages were in the stratux.log that you posted.  They wouldn't have appeared in the web UI anywhere--they just mean you didn't have your 836AI connected.

So for the 836AI, you need to connect pins 03/05 to SDA1 and SCL1 to get AHRS, and connect the USB to get GPS.  You can't leave out any of those connections.  Don't worry about SDA2/SCL2 (yes, those would be for a separate sensor).

If everything is connected properly, you should get an immediate and sensible response when you move the Stratux by hand.

boa888, 2017-08-16:
AHRS working fine tonight. Two more questions. When I first got the RY836AI I tested using USB. A cold start only took 90 seconds give or take. Now hooked up to GPIO it takes about 6 minutes. And I am looking at my tablet and I have a 3D solution but the Red GPS button under the AI is changing Green to Red. Why? It is not dependent on the number of satellites in the solution

Again, really appreciate your help

westphae, 2017-08-17:
GPS over GPIO (UART actually) isn't really working these days.  Your connection will be intermittent at best.  It's a known issue.  You'll have to connect the 836AI over USB to use its GPS for now.

closed 2017-08-17 by westphae

cyoung, 2017-08-23:
Ref #623.