cyoung/stratux #658: Wifi does not start using latest image

mikewilt, 2017-08-19:
1. Stratux version:1.4r1-f6bad36cdc.img
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [X ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I ran Stratux successfully for a long time on a Raspberry pi 2.  Eventually it stopped working and I decided to try the latest.  I downloaded the image file listed above and burned it onto an 8M SD card.  It seems to boot ok, but the wireless network never comes up.  I logged into the system and tried the "open" command, which gave me this:

root@raspberrypi:/home/pi# open

#### Stratux HOSTAPD Settings ####

* Open WiFI Option -o used.
    WiFi will be set to OPEN or UNSECURE

No errors found. Continuning...

Are you ready to apply these settings? [y/n] y
Working on /etc/hostapd/hostapd.user...
Removing WPA encryption in /etc/hostapd/hostapd.user...
Modified /etc/hostapd/hostapd.user...done

 +++ At this time the script will restart your WiFi services.
If you are connected to Stratux through the 192.168.10.1 interface then you will be disconnected
Please wait 1 min and look for the new SSID on your wireless device.
+++ Restarting Stratux WiFi Services... +++
+++ SSH will now  disconnect if connected to http://192.168.1.10 ... +++
Killing hostapd...
hostapd: no process found
hostapd-edimax: no process found
Killed...

Killing DHCP Server...

Killed...

ifdown wlan0...
ifup wlan0...
Calling Stratux WiFI Start Script(stratux-wifi.sh)...
hostapd: no process found
hostapd-edimax: no process found
hostapd-edimax-alt: no process found
hostapd-edimax-newest: no process found
Configuration file: /tmp/hostapd.conf
nl80211: Driver does not support authentication/association or connect commands
nl80211 driver initialization failed.
hostapd_free_hapd_data: Interface wlan0 wasn't started

I have seen a few references to this exact issue.  It seems like the latest Stratux image has some problem with the wireless dongle.  The wireless dongle came with the VILROS kit.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

yeo423, 2017-08-27:
Not sure if this is the best place to post, but I am having the same issue.  I updated the software to v1.4r2 yesterday and my first flight today went with no issues.  During a subsequent flight I powered up the stratux and could not see its wifi signal.  I currently have it connected to the battery pack; no wifi signal and the green light is continually blinking.  

Any help getting it back operational is appreciated!  Thank you!!

If it helps, I built the dual band version with parts listed on stratux.me site.

mikewilt, 2017-10-24:
I have a little more information.  I tried using this latest image with a brand new wifi dongle and got the same results.  Finally I reloaded from the image file that I originally started with in 2015 and it works.

I would really like to know how people are making this latest version work.  I hate to give up all the recent work on Stratux just because there is a problem with the base Linux image.

Nokomis449, 2017-10-24:
Is your new wifi dongle an EdiMax brand?  That's the only one that is guaranteed to work with a Pi 2.  Support for TP-Link, etc. was removed long ago.

cyoung, 2017-10-24:
@mikewilt - nobody is making it work. There is not a problem with the base Linux image, there is a problem with ForeFlight 9.4+. Send them a message at team@foreflight.com.

mikewilt, 2017-10-24:
EdiMax, ok that's probably my problem.  I will get that brand wireless dongle.

Nokomis449, 2017-10-24:
An alternative would be to upgrade to a Pi 3, which has onboard wifi.  A little pricier at $30-$40, but it gets you up to the latest supported board.

cyoung, 2018-01-17:
@mikewilt - if you tried the EDIMAX and it still didn't work, please reopen.

closed 2018-01-17 by cyoung