cyoung/stratux #659: Night Mode

PopsActual, 2017-08-29:
1. Stratux version:1.4r2
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: GPYes

    AHRS
      - [x] yes
      - [ ] no

    power source: 

    usb cable:

3. EFB app and version: ForeFlight

    EFB platform: iOS 10.3

    EFB hardware: iPad Mini 2

4. Description of your issue:  Enhancement request.  What are the chances of doing a night mode in the UI?  Just a button that would turn off all the lights in the Pi, SDRs, and GPYes?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-08-29:
Thanks for asking this. See below, the webui looks pretty decent with inverted colors.

Settings -> General -> Accessibility -> Accessibility Shortcut (all the way at the bottom) -> Invert Colors

Then triple-tap the home screen button to activate/deactivate.

[image]

[image]

PopsActual, 2017-08-29:
Does that turn off the hardware lights?  That's really my problem right now.

cyoung, 2017-08-29:
No, it doesn't.
Lights on the SDRs - which ones do you have? Recent ones don't have LEDs.
GPYes - can't turn those off in software.
We'd be looking at only turning off the LEDs on the Pi.

ijessen, 2017-10-17:
@cyoung are we sure the GPYes can't be turned off in software? When I looked into the u-blox 7 spec a few weeks ago it seems like a CFG-TP5 serial message (ID 0x06 0x31) could be formatted to deactivate the default time pulse output, which I believe is wired to the PPS LED on the GPYes.  I don't have anything to test on at the moment, but I would guess this should work?

cyoung, 2017-10-17:
@ijessen - you're right. Just tried this with u-center and that did it.

[image]

cyoung, 2017-10-17:
Okay, added `NightMode`. If someone wants to add something to the webui to toggle `globalStatus.NightMode`, that should round it out. Default state on led0 is "off". It will still flash if there's a critical system error.

cyoung, 2018-01-17:
We've turned off everything we can. The power LED on the Pi can't be turned off. I've had the power LED removed from the GPYes and turned ACT off with above u-blox configuration.

closed 2018-01-17 by cyoung