cyoung/stratux #662: No GPS or Weather

TArrowJim, 2017-08-31:
1. Stratux version: v1.4r1 (f6bad36cdc)
2. Stratux config: Raspberry Pi 3, Stratux 1090ES & UAT, GPYes

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no

    power source: EasyAcc

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Foreflight v9.2.1

    EFB platform: (e.g., iOS 9.2) iOS 10.3.3

    EFB hardware: (e.g., iPad Mini 2) iPad MD511LL/A

4. Description of your issue:

Just put my stratux together and the 1090/traffic reception seems fine. I see nothing on the 978/weather or GPS side. UAT on Status page shows 0 for signal and zeros across the board. GPS likewise, I see nothing. The unit is up against a window.

Log is below.

Thoughts on both 978 and GPS?

Jim

2016/02/26 01:18:08 Stratux v1.4r1 (f6bad36cdcb81e877e4dfe6e55c840a904cade0b) starting.
2016/02/26 01:18:08 Entered uatReader() ...
2016/02/26 01:18:08 read in settings.
2016/02/26 01:18:08 datalog.go: Watchdog wants to START logging.
2016/02/26 01:18:08 AHRS Info: initializing new Simple AHRS
2016/02/26 01:18:08 client connected: 192.168.10.10:4000 (iPad).
2016/02/26 01:18:08 datalog.go: dataLog() started
2016/02/26 01:18:08 AHRS Info: couldn't start ahrswebListener: dial tcp [::1]:8000: getsockopt: connection refused
2016/02/26 01:18:09 ===== ES Device Serial: stratux:1090 PPM 0 =====
2016/02/26 01:18:09 Entered ES read() ...
2016/02/26 01:18:09 Executed /usr/bin/dump1090 successfully...
2016/02/26 01:18:09 ===== UAT Device Name  : Generic RTL2832U OEM =====
2016/02/26 01:18:09 ===== UAT Device Serial: stratux:978=====
2016/02/26 01:18:09 	GetTunerType: RTLSDR_TUNER_R820T
2016/02/26 01:18:09 	SetTunerGainMode Successful
2016/02/26 01:18:09 	SetTunerGain Successful
2016/02/26 01:18:09 	GetTunerGain: 480
2016/02/26 01:18:10 	SetSampleRate - rate: 2083334
2016/02/26 01:18:10 	GetSampleRate: 2083334
2016/02/26 01:18:10 	GetXtalFreq - Rtl: 28800000, Tuner: 28800000
2016/02/26 01:18:10 	SetXtalFreq - Center freq: 28800000, Tuner freq: 28800000
2016/02/26 01:18:10 	SetCenterFreq 978MHz Successful
2016/02/26 01:18:10 	GetCenterFreq: 978000000
2016/02/26 01:18:10 	Setting Bandwidth: 1000000
2016/02/26 01:18:10 	SetTunerBw 1000000 Successful
2016/02/26 01:18:10 	ResetBuffer Successful
2016/02/26 01:18:10 	GetFreqCorrection: 0
2016/02/26 01:18:10 	SetFreqCorrection 0 Successful
2016/02/26 01:18:10 Entered UAT read() ...
2016/02/26 01:18:12 BMP280 Connection Attempt 1
2016/02/26 01:18:12 BMP280 Connection Attempt 2
2016/02/26 01:18:12 BMP280 Connection Attempt 3
2016/02/26 01:18:12 BMP280 Connection Attempt 4
2016/02/26 01:18:12 BMP280 Connection Attempt 5
[... 339 lines cut ...]
2016/02/26 01:19:40 BMP280 Connection Attempt 4
2016/02/26 01:19:40 BMP280 Connection Attempt 5
2016/02/26 01:19:40 BMP280 Connection Attempt 1
2016/02/26 01:19:40 BMP280 Connection Attempt 2
2016/02/26 01:19:40 BMP280 Connection Attempt 3
2016/02/26 01:19:40 BMP280 Connection Attempt 4
2016/02/26 01:19:40 BMP280 Connection Attempt 5
2016/02/26 01:19:40 AHRS Info: couldn't initialize BMP280 or BMP180
2016/02/26 01:19:40 AHRS Info: attempting IMU connection.
2016/02/26 01:19:40 MPU9250: Error reading calibration data from /etc/mpu9250cal.json: open /etc/mpu9250cal.json: no such file or directory
2016/02/26 01:19:40 AHRS Error: couldn't initialize an IMU
2016/02/26 01:19:44 BMP280 Connection Attempt 1
2016/02/26 01:19:44 BMP280 Connection Attempt 2
2016/02/26 01:19:44 BMP280 Connection Attempt 3
2016/02/26 01:19:44 BMP280 Connection Attempt 4
2016/02/26 01:19:44 BMP280 Connection Attempt 5
2016/02/26 01:19:44 BMP280 Connection Attempt 1
2016/02/26 01:19:44 BMP280 Connection Attempt 2
2016/02/26 01:19:44 BMP280 Connection Attempt 3
2016/02/26 01:19:44 BMP280 Connection Attempt 4
2016/02/26 01:19:44 BMP280 Connection Attempt 5
2016/02/26 01:19:44 AHRS Info: couldn't initialize BMP280 or BMP180
2016/02/26 01:19:44 AHRS Info: attempting IMU connection.
2016/02/26 01:19:44 MPU9250: Error reading calibration data from /etc/mpu9250cal.json: open /etc/mpu9250cal.json: no such file or directory
2016/02/26 01:19:44 AHRS Error: couldn't initialize an IMU

TArrowJim, 2017-08-31:
OK, still now 978 Weather but that is likely because I'm on the ground, right?
It took driving around for 15 minutes or so to finally get a GPS fix and once obtained it is much easier to get it from my 4th floor office.
Perhaps a heads up for new users.

closed 2017-08-31 by TArrowJim

cyoung, 2017-08-31:
Upgrade to [v1.4r2](https://github.com/cyoung/stratux/releases/tag/v1.4r2).

You're right about the towers on the ground. See here for a map: http://towers.stratux.me.