cyoung/stratux #664: Upgrade to 1.4r2 issue

RlaneC150, 2017-09-02:
1. Stratux version: 1.4r2
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [x] yes
      - [ ] no

    power source: KMASI battery pack

    usb cable:

3. EFB app and version: WingX Pro7 8.6.13.1

    EFB platform: iOS 10.3.3

    EFB hardware: iPad Pro and iPad 2

4. Description of your issue: Upgraded from 1.0r1 to 1.4r2 Stratux powered up had a steady green LED, wifi worked. Status would switch between connect and disconnect every 30 seconds not long enough for connections to hold nothing worked. Down graded to 1.2r2 everything worked except AHAR's.

Attaching logs for 1.4r2 and 1.2r2

[Log_Stratus_1.4r2.pdf](https://github.com/cyoung/stratux/files/1272260/Log_Stratus_1.4r2.pdf)
[Log_Stratux_1.2r2.pdf](https://github.com/cyoung/stratux/files/1272261/Log_Stratux_1.2r2.pdf)

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Ergonomicmike, 2017-09-26:
Not quite the same issue, but I tried sh'ing up from 1.4r1 (f6bad).to 1.4r2 (be67f5). Got the usual "Wait 60 seconds" msg and the UI went off line. But when it came back up, still showing 1.4r1. (Cleared cache on browser, cold start. No joy.)

cyoung, 2017-09-26:
@RlaneC150 - judging by that you're using the KMASHI battery pack on a dual band with AHRS, you might be having power issues whose symptoms have just presented after upgrading. When you upgrade to v1.4r2, does the red LED ever flicker or turn off?

@Ergonomicmike - fixed, there was an issue with the automatic builds that is fixed now.

Ergonomicmike, 2017-09-26:
@cyoung Tnx! Might get to chase lightning Wed night. If so, will try again then.

wfarqwark, 2017-09-28:
Had a problem similar to yours on 4r1.  Had longing on and it filled up /root or /pi. Fixed it with ssh.  Might have been "flight logs"

cyoung, 2017-10-17:
@RlaneC150 - if you're still having the issue after trying a different power source, please comment and re-open this issue.

@wfarqwark - next release should have reduced logging by default. When adding AHRS support some extra logging was added which tends to use up a lot of space.

closed 2017-10-17 by cyoung