cyoung/stratux #665: GPS not communicating

mspencer32, 2017-09-03:
1. Stratux version:1.4r2
2. Stratux config:

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: Easyacc power bank

    usb cable: 

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: I purchased the latest stratux from amazon (assembled) and everything works except the GPS. It says the Serial Port isn't communicating and the GPS is disconnected.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2017-09-03:
Perhaps plug the GPS into a computer to see if it works there. It should show up on a computer as a USB to serial device.

cyoung, 2017-09-03:
Link to the GPS you have?

Helno, 2017-09-03:
Sounds like a VK-162 they have the GMOUSE label.

cyoung, 2017-09-03:
There are actually "VK-162" generics that are not made by vkelcom, without the magnet, etc.

mspencer32, 2017-09-03:
Actually I found the problem. I did plug it into my computer and it didn’t see it. It seems there is a bad solder connection on the GPS unit. I peeled the label off revealing the magnet. I unscrewed the backplate and looked at the connections moving them around a bit and then it worked.

cyoung, 2017-09-03:
Where'd you get it?

mspencer32, 2017-09-03:
Amazon.

cyoung, 2017-09-04:
Glad you figured it out.

closed 2017-09-04 by cyoung

mspencer32, 2017-09-06:
Hello. I hate to say it buy my stratux stopped working today. This is my third day using it. I used it for two hours in the morning today and it worked fine. I then went for another flight and powered it on but no wifi signal. I came back home and payed with it trying to troubleshoot. I checked battery level and connections. Still no wifi signal. The green and red indicator LED's behaves unusually. It along with the Red LED go solid green for around 10 seconds after powering up. Then the Green LED starts flashing continuously. Soon the Red LED turns off and flashes intermittently. And of course no wifi signal. Can you help me?

Helno, 2017-09-06:
Red led going out means the power is going below 4.65 volts. 

Perhaps the wires in the GPS are shorting out and dragging the voltage down? Since that was a problem previously perhaps try booting up with the GPS disconnected.

mspencer32, 2017-09-06:
Yes I did recognize the Red LED turning off was a power issue. I had it plugged into a wall wart. I plugged it back into the battery and the Red LED stays on. No change with or without the GPS unit. The Green LED initially stays solid at startup and then 1 minute later begins flashing continuously. And no wifi signal.

mspencer32, 2017-09-06:
I plugged the unit into a monitor to see whats happening at startup. Everything is coming up OK except one thing. Some sort of I/O error toward the bottom of the page. I attached a photo.

Helno, 2017-09-06:
Green LED flashing indicates that there is a problem with the Stratux process. 

At this point you might want to just reimage the SD card unless you are familiar enough with Linux to troubleshoot further.

mspencer32, 2017-09-07:
I am very lost now. I deleted the files in the boot disk. I downloaded the latest firmware (1.4r2) and navigated to the IMG folder, then copied them to the boot disk. The Stratus doesn’t like that. It won’t boot up. What am I doing wrong?

mspencer32, 2017-09-07:
By the way I am not a computer programmer and don’t know Linix.

Helno, 2017-09-07:
The latest version of Stratux 1.4r2 is not in a format that can be written to an SD card, it is just a small update from 1.4r1. 

What  you will need to do is download the 1.4r1 image file and write it to the SD card following the instructions on stratux.me

mspencer32, 2017-09-07:
I just followed the instructions and got the unit to boot up again but its giving me exactly the same failure notification at startup. And of course no wifi.

mspencer32, 2017-09-07:
I haven’t changed anything leading up to this failure. Is it possible its hardware related?

Helno, 2017-09-07:
It is. 

Perhaps try removing everything from the pi and just booting up without everything else to find the issue. 

It might also be easier to troubleshoot on slack rather than here. https://stratuxadsb.slack.com

If you need a slack invite get it here. http://slack.stratux.me/

mspencer32, 2017-09-07:
I just checked it without anything plugged into it. Still the same situation. I’m going to have to return this to Amazon. I wasn’t prepared for extensive trouble shooting nor do I have the time. I just need something that works.

Helno, 2017-09-07:
Could just be a defective pi. If you purchased it through Amazon they will get it swapped out quickly.