cyoung/stratux #667: Cannot personalize wifi password...

Sakuranbo-ni, 2017-09-17:
I SSHed into my Stratux using PuTTy, switched to root and changed my SSID using hostapd_manager.sh. however, I cannot seem to change my password. I used the "secure" command, that at least applied encryption and set a password, the good old "SquawkDirtyToMe!" however, I want to personalize. When I run "hostapd_manager.sh -p XXXXXXXX" (not actually X's) it then asks me if im ready to apply changes, which i agree to, it runs its reboot and tells me i should see my new SSID.....however, nothing has changed. Please and thank you!!!

cyoung, 2017-09-22:
Just tried it, not working. `hostapd_manager.sh -e` works to add the default password.

cyoung, 2017-09-22:
Fixed by 7f6f139.

closed 2017-09-22 by cyoung