cyoung/stratux #669: stratux.me www pg "Pre-Built" Option 3 (OpenFLight Solutions) leads to -404

bzcameron, 2017-10-07:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

bzcameron, 2017-10-07:
AND THX for the resources! Greatly appreciated  BDC

cyoung, 2017-10-07:
Thanks! Fixed now.

commit 7cd7dd62 (2017-10-07): Fixes #669, broken OFS pre-built link.
closed 2017-10-07 by cyoung