cyoung/stratux #670: Feature request - Traffic audio alerts

cbfraser, 2017-10-12:
Generate configurable audio alerts based on traffic -distance, altitude, etc

peepsnet, 2017-10-12:
This has been requested before. At this time, it has been seen as a feature of the EFB that you are using. We are only looking to provide as much quality data/info to the connected devices allowing them to make those decisions!!!

cbfraser, 2017-10-12:
understandable but having stratux generate the audio allows an audio cable between stratux and intercom as a more permanent install rather than plugging in and removing both power and audio  when using a tablet  EFB.  And allows for audio support when the EFB doesn’t support it (eg FlightPlan GO on Android)  and perhaps best of all – 

 

allows for a custom criteria to generate better audio – eg traffic two oclock low.

 

Anyway the traffic.go code sets an alert flag based on distance and has comment – 

 

// TODO: Could be more intelligent, taking into account headings

 

Could you point me to how to contribute (in this case isTrafficAlertable)?

 

thanks

peepsnet, 2017-10-13:
https://github.com/cyoung/stratux/issues/411

https://github.com/cyoung/stratux/issues/364

cyoung, 2017-10-17:
@cbfraser - if you're interested in doing this, then we may add it. What are your abilities?

`apt-get install espeak`, the 'espeak' tool seems to work fairly easily:

```
root@raspberrypi:~# espeak --stdout "Traffic, ten o’clock, one two miles, southeast bound, one thousand feet below you" | aplay
Playing WAVE 'stdin' : Signed 16 bit Little Endian, Rate 22050 Hz, Mono
root@raspberrypi:~# 
```

(from [JO 7110.65W](https://www.faa.gov/documentLibrary/media/Order/ATC.pdf))

We won't (and don't need to) add anything to traffic.go for this, it should be an external program.
[traffic.zip](https://github.com/cyoung/stratux/files/1392289/traffic.zip)

cyoung, 2017-10-18:
By the way, we could do the similar with radar, METARs, PIREPs, etc.

https://github.com/cyoung/stratux/blob/master/test/nexrad_annunciator.go

cbfraser, 2017-10-18:
good to hear –

I’ve got a computer science background – and while it has been awhile since I coded professionally,

the coding is straight forward – I’ve done some Arduino stuff but no rp3 so would likely need pointers

on integration with traffic.go.  

I’ve done a couple of startups so pretty good a product definition, requirements, testing, usability.

You mention not changing traffic.go but the isTrafficAlertable  code sets the alert flag and even the comment suggests

improvement (and some users rebuild image to clear that code so the efb app can do it according to its criteria).

Wouldn’t it makes sense to use same logic for setting alert flag and generating audio alert?

 

I’ve got espeak running – Let me put together a proof of concept system

and see what issues there are – such as having more traffic to report than audio bandwidth,

likely configuration options,

and I’ll get back with some ideas.

 

Do you care whether audioalert is go or python?

 

 

 

func isTrafficAlertable(ti TrafficInfo) bool {
      // Set alert bit if possible and traffic is within some threshold
      // TODO: Could be more intelligent, taking into account headings etc.

cyoung, 2017-10-18:
@cbfraser - it doesn't matter if it's in Go or Python. We have Python installed on the release images.

It would make sense to use the same criteria but I think you mentioned making it customizable. That makes sense for audio alerts perhaps, but not really customizing traffic alerts in stratux. If it's in the main stratux code I'd prefer that it be based on some sort of standard and not customizable at all.

cbfraser, 2017-10-18:
makes sense –traffic would have a base   criteria for the alert flag - eg 2nm as it is now but also altitude (ie within 1,000ft) – 

and put in audioalert configurable criteria and other criteria like closure, collision detection etc

 

I’ll come back with a suggested proposal as a starting point…..

 

thanks

cbfraser, 2017-10-19:
I’ve done some preliminary work – the code

for “traffic 12 oclock high 2 miles” is straight forward based on info   available in traffic.

 

I’ve also done a draft collision prediction section – (2d)

“traffic 12 oclock high 2 miles incoming 3 minutes”   that brings up a few questions:

-          is ownship and target   track and velocity available  - I see the fields – are they usually available and valid?

-          I’m using mpmath and sympy – would it be a problem including those libraries if you wanted to use this?

 

thanks

cyoung, 2017-10-20:
Excellent. We can include those libraries, yes. See `ws://localhost/traffic` and `http://localhost/getSituation` documented [here](https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md). Sometimes that information is available, sometimes not.

cbfraser, 2017-10-24:
I have initial code for traffic alerts and collision detection –

 

do I use this email to send you discussion material?

 

regards

cbfraser, 2017-10-25:
I’ve done the straight forward traffic alert program.

I opted for a typical terse ATC traffic report –

“Traffic 12 oh clock high 5 miles”. It could be more verbose but I’m concerned that in

busy traffic audio bandwidth may be limited.

I’ve spent most of my time on collision detection which was far more complicated

and more interesting  -  will send that next.

 

I’ll need some direction on integration please– accessing the traffic.go data,

adding (better you add though) & reading  configuration parameters & logging.

 

Out for a few weeks so can discuss but will do further work on return.

 

regards

 

import subprocess

def audio_out(audio_str):
    """Generate audio out with eSpeak
    """
    # need to pass eSpeak options
    # bluetooth option
    p = subprocess.Popen(["C:\\eSpeak.exe",audio_str])
    return

 

#!/usr/bin/env python

from traffic_alert import traffic_alert
from audio_out import audio_out

while 1==1:
    b = int(input("bearing: (90)") or 90)
    v = int(input("vertical: (2000)") or 2000)
    m = int(input("miles: (5)") or 5)

    print(traffic_alert(b,v,m))
    audio_out(traffic_alert(b,v,m))

#!/usr/bin/env python

import math
from math import atan, degrees
 
#convert ft to nm
FT2NM = 0.0001645784

#possible configuration values
high_angle = 15 # degrees
low_angle = -15 # degrees

def traffic_alert(bearing, vertical, distance):
    """Generate audio alert string
    """
    #  bearing is relative to heading (-180 :: +180)
    # vertical: feet +- relative to ship
    # distance in nm
    
    if bearing > 0:
        oclock = int((bearing + 15) / 30)
    else:
        oclock = 12 + int((bearing - 15) / 30)
    if oclock == 0:
        oclock = 12
        
    #calcuate visual angle
    angle = math.degrees(math.atan(vertical*FT2NM/distance))
    visual_angle = ""
    if angle < low_angle:
        visual_angle = "low "
    elif angle > high_angle:
        visual_angle = "high "
    
    return ("traffic " + str(oclock) + " oh clock " + visual_angle + str(int(distance))
            + " mile"+[" ","s, "][int(distance)>1])

cyoung, 2017-10-25:
@cbfraser - I see some Python code, appears to be three scripts added to the body of the message. Yes, please use this thread to track progress. Could you maybe add the files to a ZIP and post on GitHub. I'll go from what you have to add the piece that pulls traffic information from Stratux and you can expand on it from there.

cyoung, 2017-10-25:
@cbfraser - committed these scripts to:

https://github.com/stratux/traffic_alert

See also the script that I added called "get_traffic.py". It will print traffic and situation updates to the screen. Perhaps you could fork from this and add changes on your GitHub account.

cbfraser, 2017-10-25:
Here is the more complex collision code - the test driver has option to output aircraft tracks to plotly (helpful when trying to visualize different scenarios) and optionally will read file with test data.
Requires math and sympy and also plotly & plotly account if plotting output.
collision_test.py - test driver
collide.py  - collision code
Capture.jpg - sample plot
test data template.tst - input values for test data
headon.tst - 9 head on scenarios
intersection.tst - 5 intersecting scenarios

Please post if a test scenario reveals an error (ideally with *.tst file).
[collision.zip](https://github.com/cyoung/stratux/files/1416361/collision.zip)

At a high level - traffic would be filtered on distance and vertical seperation,
then traffic_alert would generate string for traffic and collision add collision warning if detected.

cbfraser, 2017-11-24:
ready to start integration testing - I've done python code in a windows environment, can I use the target rpi3 for development or do I need a 2nd one or a cross compile environment?

Nokomis449, 2017-11-24:
The Raspberry Pi image includes Python (I believe 2.7).  I'm developing on the Pi by SSH'ing in from a Windows 10 machine to run the code, and using WinSCP and Code to edit the scripts on the Windows 10 machine rather than use an editor on the Pi.  I'm fairly new to Python and this setup works well for me.

cbfraser, 2017-11-30:
integration on target rpi3 update:
using 2nd board with ethernet for intial setup
stratux install
audio & espeak
sympy
websocket
test programs 
traffic & collision alerts

now need to turn on traffic demo & rebuild stratux for bench testing

cbfraser, 2017-12-03:
Configuration Parameters:
Need to define user/system parameters:
Filter for audio alerts based on:
traffic_distance
traffic_separation (vertical)
collision_window
audio pacing for same target
angle for target high/low

cbfraser, 2017-12-04:
first audio reports generated from live traffic -
error in data feed - I'll post separately

cyoung, 2017-12-04:
Did you have record logs enabled? Could you share the logs?

cbfraser, 2017-12-04:
yes I can help you with that - what files do you want?

2017/12/04 17:27:28 C04414 => {"Icao_addr":12600340,"Reg":"C-FZUH","Tail":"e C-FZUH","Emitter_category":0,"OnGround":false,"**_Addr_type":0_**,"TargetType":0,"SignalLevel":-28.77129077135564,"Squawk":0,"Position_valid":false,"Lat":0,"Lng":0,"Alt":10475,"GnssDiffFromBaroAlt":0,"AltIsGNSS":false,"NIC":0,"NACp":0,"Track":0,"Speed":0,"Speed_valid":false,"Vvel":0,"Timestamp":"2017-12-04T17:27:26.76Z","PriorityStatus":0,"Age":1.74,"AgeLastAlt":1.42,"Last_seen":"0001-01-01T00:02:14.14Z","Last_alt":"0001-01-01T00:02:14.46Z","Last_GnssDiff":"0001-01-01T00:00:00Z","Last_GnssDiffAlt":0,"Last_speed":"0001-01-01T00:00:00Z","Last_source":1,"ExtrapolatedPosition":false,**_"BearingDist_valid":true_**,"Bearing":63.815863409600276,"Distance":1.2317282764558565e+07}

cyoung, 2017-12-04:
/var/log/stratux.sqlite . Enable all logging in the webui first.

cbfraser, 2017-12-04:
[stratux.zip](https://github.com/cyoung/stratux/files/1528236/stratux.zip)
here it is
renamed from .sqlite to .txt to access from windows

cbfraser, 2017-12-04:
also seeing these same static values on type 1 targets (adsb)
{
  "Icao_addr": 12615599,
  "Reg": "C-GWJE",
  "Tail": "eaC-GWJE",
  "Emitter_category": 0,
  "OnGround": false,
  "Addr_type": 0,
  **_"TargetType": 1,_**
  "SignalLevel": -27.09742730605482,
  "Squawk": 0,
  "Position_valid": false,
  "Lat": 0,
  "Lng": 0,
  "Alt": 20925,
  "GnssDiffFromBaroAlt": 325,
  "AltIsGNSS": false,
  "NIC": 8,
  "NACp": 8,
  "Track": 82,
  "Speed": 384,
  "Speed_valid": true,
  "Vvel": 3328,
  "Timestamp": "2017-12-04T23:23:09.808Z",
  "PriorityStatus": 0,
  "Age": 26.03,
  "AgeLastAlt": 20.63,
  "Last_seen": "0001-01-01T00:58:52.36Z",
  "Last_alt": "0001-01-01T00:58:57.76Z",
  "Last_GnssDiff": "0001-01-01T00:58:52.91Z",
  "Last_GnssDiffAlt": 20650,
  "Last_speed": "0001-01-01T00:58:52.91Z",
  "Last_source": 1,
  "ExtrapolatedPosition": false,
  "BearingDist_valid": true,
  "Bearing": 63.815778004276424,
  "Distance": 1.231728205534869e+07
**_earingDist_valid": true,
  "Bearing": 63.815778004276424,
  "Distance": 1.231728205534869e+07_**
}

cbfraser, 2017-12-04:
Reg and Tail values seem to always be the same

{"Icao_addr":12594428,"Reg":"C-FRAX","Tail":"e C-FRAX","E.....

I believe reg or tail is usually set to flight number on commercial flights,
so could be an issue

while planefinder.net is showing flight number WS713 - C-FRAX

cbfraser, 2017-12-16:
> Audio alerts would be really great because it would help to overcome integration probs with nav-software that does not (yet) compute the stratux output. Consider it an excellent feature.
> Since I am no good at programming I'd like to assist any testing on ground and in flight.

good to hear

I think the other benefit is not having to be locked on watching the EFIS and also not having to estimate what traffic is a threat.

 

Do you know how to rsh to Stratux and copy files to the Stratux?

 

Do you pick up ADSB traffic at your house?

 

Does your aircraft have ADSB out?

 

Do you have a friend willing to help with flight testing with an aircraft with ADSB out?

peterclubroad, 2017-12-17:
Two planes (one being equiped with a mode S transponder already. I am about to put in a gps module to activate extended squitter.) incl friends are available. PCB for that finished yesterday. Next airfield is less than 10 miles :-)
And yes, I have fiddeled about with ADSB trafic reception for a while. I am am not continuously feeding it to the web though. Copying files to stratux/pi3 has worked, too.
Let me know if you think it's worth a try.

cbfraser, 2017-12-17:
good – I’ll prepare some setup instructions and test files

and post later

get started with:

sudo apt-get install espeak

sudo pip install sympy

sudo pip install websocket-client

 

in case Chris has the inclination a build with these packages would be very helpful

CraXgt, 2018-01-01:
What’s the progress on this one? I also consider this a killer feature for a permanent install which is independent from the software used.
I can only offer my help testing with three planes, two ads-b out equipped.

Patrick

cbfraser, 2018-01-02:
I’ve got a   solution running on the bench picking up airborne traffic –

it does basic traffic alerts (“traffic 12 oh clock high 2 miles”)  and can

predict collisions (“ incoming in 2 minutes”). 

And has a simple webpage with status:

 

 

As I get time I need to do some work to package and document it.

 

As well the Stratux build needs to have espeak, backports, websocket, mpmpath and sympy installed.

Perhaps Chris could help by releasing a build with these packages.

 

Meanwhile I will work to document package installs and confirmation tests,

and then audio_alerts.py  itself: install, setup, usage and suggested test regime.

I’d recommend pilot and operator in one aircraft with operator monitoring Stratux and

providing commands to 2nd ADSB out aircraft.

 

Are you familiar with installing packages on rpi3?

 

And for convenience if someone would take on testing and documenting Bluetooth with the Stratux

build that would be very helpful.

CraXgt, 2018-01-02:
Sounds good - I'm familiar with installing additional packages on the Pi3. A build including these and your changes would be appreciated. It could also potentially get more testers involved for this feature.

Maybe add an optional audio message on startup, calling out the correct bootup process of stratux like "Stratux system check okay" or similar - and in case of a error on startup another phrase informing the pilot. It'll also confirm the correct connection of the audio output to the intercom.

looking forward to testing.

Patrick

cbfraser, 2018-01-02:
Good
It would be ideal to have Chris include these packages in current buildBut I'll see if I can make an interim one.Easy to add my python script to the build.
Script does generate a start-up audio and also a periodic status message to provide feedback.
My email is on my GitHub profile send me a message and we'll take the details offline and I'll keep posting Key Parts here
Cameron Fraser 604-500-4682

cbfraser, 2018-01-10:
Still a work in progress, getting closer to field testing but here's an overview

Audio Alerts:

Audio alerts is set of python scripts that runs on the Stratux build and reads all the 

traffic messages on the websocket interface then generates audio traffic and collision 

alerts. It was made possible by Chris providing a python hook to his Stratux build.

Requirements:
As well as Stratux - mpmath, sympy, websocket, backports, espeak and/or omxplayer

Operation:

Audio alerts initially filters traffic messages based on target's altitude, distance, 

ground status and ship's GPS status.

After that it tracks that traffic and optionally filters on whether traffic is getting 

closer and optionally calcuates potential collision.

Audio is paced at a system output rate and a rate for a target
and a periodic audio heartbeat is issued.

Traffic alerts are of the form "Traffic 12 oh clock high 6 miles"

Collision alerts are filtered by a near miss time window but disregards relative 

altitudes. It will predict ship & target crossing headon or from behind or at an 

intersection.

Collision alerts are of the form "Headon  in 1 minute", "Passing  in 1 minute", 

"Overtaking  1 minute", "Contact in 1 minute"

Audio is either espeak rather robotic but complete or omxplayer which uses prerecorded 

voice. This sounds better but is limited to 20 miles and 12 minutes. It does allow for 

other languages.

A simple webpage displays current traffic and stats. 192.168.10.1/audio.html

Bluetooth not supported - hopefully someone will document setup.

cbfraser, 2018-02-21:
audio alerts is ready for some initial field testing - need 2 aircraft - one with adsb out,  one with the stratux and observer to run tests.....

cbfraser, 2018-09-05:
> Hi cbfraser, how is this doing?  Did you ever get any responses on test flights?  There was a discussion on the slack discussion channel about having voice traffic notifications yesterday and a couple off us where curious.
>
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/cyoung/stratux/issues/670#issuecomment-418695321",
> "url": "https://github.com/cyoung/stratux/issues/670#issuecomment-418695321",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> },
> {
> "@type": "MessageCard",
> "@context": "http://schema.org/extensions",
> "hideOriginalBody": "false",
> "originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
> "title": "Re: [cyoung/stratux] Feature request - Traffic audio alerts (#670)",
> "sections": [
> {
> "text": "",
> "activityTitle": "**cobraPA**",
> "activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
> "activitySubtitle": "@cobraPA",
> "facts": [
>
> ]
> }
> ],
> {
> "name": "Add a comment",
> "@type": "ActionCard",
> "inputs": [
> {
> "isMultiLine": true,
> "@type": "TextInput",
> "id": "IssueComment",
> "isRequired": false
> }
> ],
> "actions": [
> {
> "name": "Comment",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"cyoung/stratux\",\n\"issueId\": 670,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
> }
> ]
> },
> {
> "name": "Close issue",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"cyoung/stratux\",\n\"issueId\": 670\n}"
> },
> {
> "targets": [
> {
> "os": "default",
> "uri": "https://github.com/cyoung/stratux/issues/670#issuecomment-418695321"
> }
> ],
> "@type": "OpenUri",
> "name": "View on GitHub"
> },
> {
> "name": "Unsubscribe",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 266129812\n}"
> }
> ],
> "themeColor": "26292E"
> }
> ]

Several promises but no one came through...

Cameron Fraser604 500 4682

PatrickDCarter, 2018-09-05:
I was the one asking on Slack about the progress, I’ve got a plane and a stratux, but not a ton of time, I could test when I can if you need.

cbfraser, 2018-09-05:
For effective/repeatable testing 2 aircraft are needed – 2 pilots and one tester.

 

Several packages need to be included in the base build, I haven’t seen any interest

in including these.  I’ve got a one off build that I’ll have to dust off if we can get

test aircraft.

 

And Bluetooth needs to be configured.

 

Overview below

…

Audio Alerts:

Audio alerts is set of python scripts that runs on the Stratux build and reads all the

traffic messages on the websocket interface then generates audio traffic and collision

alerts. It was made possible by Chris providing a python hook to his Stratux build.

Requirements:
As well as Stratux - mpmath, sympy, websocket, backports, espeak and/or omxplayer

Operation:

Audio alerts initially filters traffic messages based on target's altitude, distance,

ground status and ship's GPS status.

After that it tracks that traffic and optionally filters on whether traffic is getting

closer and optionally calculates potential collision.

Audio is paced at a system output rate and a rate for a target
and a periodic audio heartbeat is issued.

Traffic alerts are of the form "Traffic 12 oh clock high 6 miles"

Collision alerts are filtered by a near miss time window but disregards relative

altitudes. It will predict ship & target crossing head on or from behind or at an

intersection.

Collision alerts are of the form "Head on in 1 minute", "Passing in 1 minute",

"Overtaking 1 minute", "Contact in 1 minute"

Audio is either espeak rather robotic but complete or omxplayer which uses prerecorded

voice. This sounds better but is limited to 20 miles and 12 minutes. It does allow for

other languages.

A simple webpage displays current traffic and stats. 192.168.10.1/audio.html

Bluetooth not supported - hopefully someone will document setup.

 

 

 

Cameron Fraser

+1 604 500 4682

PatrickDCarter, 2018-09-05:
I’m sure I can get a few airplanes, that shouldn’t be a problem.

cbfraser, 2018-09-05:
ok – I will send test plan for you to review.

 

To be really useful someone needs to document how to configure Bluetooth.

 

 

 

Cameron Fraser

+1 604 500 4682

cbfraser, 2018-09-05:
attached

 

Cameron Fraser

+1 604 500 4682

 

Flight Tests V1.0:

Recommend aircraft A -flown 2 people, copilot monitoring Stratux
and giving commands to aircraft B and taking notes so pilots are always 
able to maintain safe seperation - B always 1000ft below A except for tests #3-6.

need laptop to be able to ssh to Stratux to monitor traffic webpage and  edit parameter files
and restart. -  

notepad for results & comments about each test

use "script" to start logging terminal session
and at the end "exit" to stop - email file typescript and also
 notes

Parameters - edit audio_alerts.txt:
vertical 2000
distance 10
converging traffic False
collision 5
simulator False
voice directory - null or en-GB

put aircraft B in the circuit
#1:during taxi no reports

#2:on takeoff roll reports should start 

scenario 1:
A -straight and level
B - starting at 3 oclock and at distance of about 1/2 mile  steady 
	(ie "wingman" position")
	 - #3:check traffic reports.
	- turn on coverging traffic filter True
		#4:check reports stop (B constant or increasing distance)
		have aircraft B approach (slowly!)
		#5:check reports start
		turn off converging filter
	- aircraft B climbs until report is "Traffic x oh clock HIGH"
		#6: check angle - too soon too late?
	- aircraft B descends until report is "LOW"
		#7:check angle - too soon too late?
	- aircraft B resumes 1000 ft below A , 1 mile, 3 oclock and 		circles A  horizontally until in 12 oclock about 1 or 2 nm ahead
			check  collision alerts 3 cases:	 
			-#8:overtaking inside window - B much slower
			-#9:overtaking outside window - B slight slower
			-#10:not overtaking - B faster
	- B goes in front 20 miles and 1000ft below A and does 180,
	 ie headon - careful! - check distance on webpage, 20 miles
		should give about 6 minutes if each aircraft is doing
		about 100kts
		#11:check traffic warnings"head on in x minutes" reducing
		 down to #12:"head on now!"

	- B moves past 1000ft below to 6 oclock - does 180 about 1 nm 
	  behind.
			- #13: passing outside window - B slightly faster
			- #14: not passing - B slower
			- #15: passing inside window - B faster
			- #16: B changes track 15 degrees
				-traffic 6 oclock  but not passing
	-B moves to 3 oclock 1000ft lower,  1 mile, out parallel tracks
		- changes track 10 degress left of A
			- #17:intersection but not on extended track
		- changes track to 45 degrees left of B
			- #18: contact in x minutes 

		 
	- try  other voice ,which is better in cockpit?

cobraPA, 2018-09-06:
> Several packages need to be included in the base build, I haven’t seen any interest
> 
> in including these.  I’ve got a one off build that I’ll have to dust off if we can get
> 
> test aircraft.
> 

I suspect packages will not be added to the main build until testing is done, they would just be extra and unused.  This adds risk to the release until something that uses them is validated.  (Not speaking for the project, but I suspect that's why they haven't been added yet.)

> 
> 
> And Bluetooth needs to be configured.

What is bluetooth used for?  What connections should it support?

cbfraser, 2018-09-06:
that makes sense

 

Bluetooth is to get audio to headset – eg Lightspeed – otherwise need to run an audio cable either to headset (cumbersome)

or audio panel which might not be available….

so convenient but not essential I guess

 

 

 

Cameron Fraser

+1 604 500 4682

cobraPA, 2018-09-13:
I'm working on an image for my own use, I may be able to post something you could use as well.

Another question, I'm curious about your traffic algorithm.  I'd like to test traffic notices on a local OLED display.  Is it all your own custom code?  How are you doing the proximity detection?

cbfraser, 2018-09-13:
Yes my code but cyoung did the heavy lifting putting in hooks that I used…

 

 

Its not about proximity – it is about potential path intersection

 

A nearby aircraft headed away is of no interest..

 

 

Simple linear geometry -   with speed and direction known you can calculate intersection point and arrival times of each aircraft at that point

 

and there are special cases  for head on collision  , overtaking and overtaken.

 

I can post code when we get some testing done –

the key package to use mpmath for trig stuff  and sympy for solving linear equations (amazing package)

I can post the key bit if you like – the bulk of it is just lots of filtering and scenario checking….

 

 

I also used plotly and did plots of intersection for my bench testing

 

regards

 

Cameron Fraser

+1 604 500 4682

d-c-d, 2018-09-13:
I'd like to help if I could.

I fly out of Paine Field in Washington - any other stratux developers that
need test flights?

David

cbfraser, 2018-09-13:
My testing needs 2 aircraft,, 3 people....

Cameron Fraser604 500 4682

"activityTitle": "**d-c-d**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@d-c-d",

saabnut, 2019-04-18:
Where are you guys on this. I am in south florida and have an ads-b out aircraft, and could round up the others needed for test flights.

cbfraser, 2021-01-27:
I've revived this because I'm planning on upgrading my aircraft to ADSB and there is enough ADSB traffic near by to do bench testing.   If anyone interested in flight testing I can post current code. Not packaged so need to be technical enough to flash, ,ssh etc.
 
[image]

saabnut, 2021-01-27:
Is there any need to change hardware, or do I plug stratux into aux input
for my intercom? If simply updating flash, I will be willing to flight
test.
I do not want alerts at 6NM however. Is a filter implemented, and
adjustable.

cbfraser, 2021-01-27:
no h/w change – I’m using audio port although if someone could configure

Bluetooth that would be nice.

 

It is more involved that just flashing an image but yes configurable

for distance, vertical separation and collision….

 

With some more testing I may create a complete image.

 

Audio Alerts:

 

Audio alerts is set of python scripts that runs on the Stratux build and reads all the traffic messages on the websocket interface and generates audio traffic and collision alerts. 

It was made possible by Chris providing a python hook to his Stratux build.

 

Requirements:

As well as Stratux - mpmath, sympy, websocket, backports, espeak and/or omxplayer

 

Operation:

 

Audio alerts initially filters traffic messages based on target's altitude, distance, ground status and ship's GPS status.

 

After that it tracks that traffic and optionally filters on whether traffic is getting closer and optionally calcuates potential collision.

 

Audio is paced at a system output rate and a rate for a target.

 

Traffic alerts are of the form "Traffic 12 oh clock high 6 miles"

 

 

Collision alerts are filtered by near miss time window but disregards relative altitudes. It will predict ship & target crossing headon or from behind or at an intersection.

 

Collision alerts are of the form "Headon  in 1 minute", "Passing  in 1 minute", "Overtaking  1 minute", "Contact in 1 minute"

 

Audio is either espeak rather robotic but complete or omxplayer which uses prerecorded voice. This sounds better but is limited to 20 miles and 12 minutes. It does allow for other languages.

 

A simple webpage displays current traffic and stats. 192.168.10.1/audio.html

 

Bluetooth not supported - hopefully someone will document setup.

 

Defects & suggestions  c@cbfraser.ca

 

Cameron Fraser

+1 604 500 4682

saabnut, 2021-01-27:
sounds great. too many items to watch these days.

cbfraser, 2021-03-04:
Here is an image that starts audio alerts on boot.  Burn image and connect to audio jack.
Configure by editing audio_alerts.txt and use winscp(windows) or andFTP(android) to copy that file from/to theStratux. Zip file is an image from 16GB SD card.
https://drive.google.com/drive/folders/1g9YYz7hIjP7-5d-PSo6aqaohBuIIs3Ur?usp=sharing
[audio_alerts.txt](https://github.com/cyoung/stratux/files/6086778/audio_alerts.txt)
[overview.txt](https://github.com/cyoung/stratux/files/6086779/overview.txt)

For anyone intestested in testing with a 2nd aircraft that has ADSB out I have a suggested test plan.