cyoung/stratux #671: No wifi with jump from 1.2r2 to 1.4r3

glongpilot, 2017-10-14:
1. Stratux version: 1.4r3
2. Stratux config: dual SDR, Stratux GPS, AHRS, external power supply, RPi 3.
3. EFB app (Avare). EFB platform (Android). EFB hardware (Nexus 10).
4. Description of my issue. Wifi does not work. With 1.2r2 in the same configuration the wifi works fine. Tried burning the 1.4r3 twice (onto 16 GB micro SD) to be sure it burned okay. Then burned 1.2r2 on separate 4 GB micro SD to test. 1.2r2 was previously running on the 16 GB micro SD.

n444pf, 2017-10-16:
Also no wifi with 1.4r3

cyoung, 2017-10-16:
@glongpilot @n444pf - what program are you using to burn the image?

glongpilot, 2017-10-16:
I used Win32 Disk Imager version 0.9.5.

cyoung, 2017-10-16:
Were you sure to check that you were using the .img file and not the .img.zip file?

Try with http://etcher.io.

glongpilot, 2017-10-16:
I must not have at first. I may have used a folder (via the icon for the file) with the .img extension. However I took special care to redo it again just now after moving the unzipped .img file to another location and it does now work. Thank you!

cyoung, 2017-10-16:
Excellent, thanks for giving it another try.

@n444pf - would you perhaps have a similar issue?

cyoung, 2017-10-17:
Post back if issue persists, @n444pf.

closed 2017-10-17 by cyoung