cyoung/stratux #672: OLED support is out-of-date

Snowflake6, 2017-10-16:
1. Stratux version: 1.4r3
2. Stratux config:

    SDR
      - [ ] dual low pwer

    GPS
      - [ ] yes - VK172 right now, but have a GPYes on order

    AHRS
      - [ ] have used in the past and have AHRS/Fan Control board on order

    power source:
      - Wall adapter, 2A cigarette lighter adapter

    OLED Screen:
      - DIY Mall 0.96" 128x64
        https://www.amazon.com/gp/product/B00O2LKEW2/

3. Not relevant

4. Description of your issue:

The structure of the LUMA OLED drivers was changed in January but the Stratux code hasn't been updated to reflect the change.  A fresh burn of 1.4r3 onto a new SD card is completely non-functional.

Request:
1. Update the OLED drivers
     https://luma-oled.readthedocs.io/en/latest/api-documentation.html
2. Update the scripts driving the OLED screen to reflect the new drivers
     https://www.reddit.com/r/stratux/comments/61grto/assistance_with_oled_screen/
     
I'm currently building up a new unit from a fresh 1.4r3 install, and will document each step and add to this issue.

Snowflake6, 2017-10-16:
From previous work:
---
The upgrade is pretty straightforward:

- Installation of luma.oled: https://luma-oled.readthedocs.io/en/latest/install.html
```
sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg-dev
sudo -H pip install --upgrade pip
sudo apt-get purge python-pip
sudo -H pip install --upgrade luma.oled
```

- Change calls to OLED driver: https://luma-oled.readthedocs.io/en/latest/api-documentation.html
- in stratux-screen.py, replace:
```
from oled.serial import i2c
from oled.device import ssd1306, sh1106
from oled.render import canvas
```
- with:
```
from luma.core.serial import i2c
from luma.oled.device import ssd1306, sh1106
from luma.core.render import canvas
```
Note: Do NOT replace oled.width calls with luma.core.width as suggested in the link above. (oled.width isn't a call to the driver, it's a reference to the width property of the oled variable.)

Snowflake6, 2017-10-16:
Not enabling any of this by default (as the number of users is currently low) is okay, but loading it requires getting an internet connection to the Pi, which is difficult.  If it was there to work with, one could ssh in and enable it, at least.

Snowflake6, 2017-10-17:
Adding stratux-screen.py that works after installing new OLED drivers as above.

[stratux-screen.py.zip](https://github.com/cyoung/stratux/files/1390004/stratux-screen.py.zip)

cyoung, 2017-10-17:
I run into an issue with the second command, how did you get around it?

```# pip install --upgrade luma.oled
Downloading/unpacking luma.oled
  Downloading luma.oled-2.2.11-py2.py3-none-any.whl
Downloading/unpacking luma.core>=1.0.0 (from luma.oled)
  Downloading luma.core-1.0.2-py2.py3-none-any.whl (49kB): 49kB downloaded
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1266, in prepare_files
    req_to_install.extras):
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2401, in requires
    dm = self._dep_map
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2597, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2630, in _compute_dependencies
    common = frozenset(reqs_for_extra(None))
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2627, in reqs_for_extra
    if req.marker_fn(override={'extra':extra}):
  File "/usr/lib/python2.7/dist-packages/_markerlib/markers.py", line 113, in marker_fn
    return eval(compiled_marker, environment)
  File "<environment marker>", line 1, in <module>
NameError: name 'platform_system' is not defined
```

Snowflake6, 2017-10-17:
It worked for me when run as
>sudo -H pip install --upgrade luma.oled

Snowflake6, 2017-10-17:
The original instructions for the step of installing luma.oled have an apt-get install and apt-get purge that I was thinking may not be needed.  Running the apt-get install resulted in a message that all four packages were already installed and up-to-date on my fresh 1.4r3 build.  Since they were already there, I didn't see the point in purging the python-pip package.  So while I ran all four commands when I did my build, I re-installed python-pip after everything was done just in case something else needed it.

>$ sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg-dev build-essential
>$ sudo -H pip install --upgrade pip
>$ sudo apt-get purge python-pip
>$ sudo -H pip install --upgrade luma.oled

cyoung, 2017-10-17:
Did it on a new v1.4r3 image, actually the master image in a chroot environment.
`easy_install -U pip` did it for the pip upgrade.

Should be working on the next image.

closed 2017-10-17 by cyoung

Nokomis449, 2017-10-17:
Does this mean the next image will drop the old oled drivers and contain the new ones?  I haven't updated my personal screen script and it's been working all along with the old drivers.  Sounds like I'll need to upgrade.  If it's inevitable, I might as well do it now while I'm rewriting it.

Snowflake6, 2017-10-17:
@Nokomis449, I included an updated stratux-screen.py above that works with 1.4r3 and the latest OLED drivers.  I've been working on a HowTo to get the OLED screen working but may abandon that if the next image will have the support there by default anyway.

Nokomis449, 2017-10-17:
Yeah I saw that, but I don't use stratux-screen.py.  My own screen script still uses the old drivers.  I guess this is the push I needed to update to the new ones.

Snowflake6, 2017-10-17:
@Nokomis449, the only change you should need is the change to update the calls to load the drivers, as in the "From Previous Work" post above.  The function calls beyond that are the same for both versions.

Snowflake6, 2017-10-17:
@Nokomis449, if you could post your latest script I would be happy to look at updating it for you... I've been meaning to ask you for it anyway as you have multiple screens working there and I'm going to look at how hard it would be to get an arrow pointing to the closest target on screen.  Your code may be useful in figuring out how to do that.

Nokomis449, 2017-10-17:
[screenfan.zip](https://github.com/cyoung/stratux/files/1392419/screenfan.zip)

Here's my latest but it's about to change drastically so no need to make changes for my sake; it'll probably be a total re-write.  Assuming they integrate with the Pi ok, I'm about to add a CO detector, lightning detector, compass, outside temp/humidity, ship's voltage meter, and a couple of momentary buttons/lights to indicate alert status.  The screen script will be updated to display all that data in a user-selectable format using one of the momentary switches.

Snowflake6, 2017-10-17:
Wow, that's quite an ambitious rewrite... You might want to look at a larger screen!  :)

Thanks for the heads-up on the fan PWM.  I guess I can either disable the stock one or the one in your script, with a little effort.  I'll read carefully before trying anything.

Nokomis449, 2017-10-21:
I just re-read your last comment.  You can easily disable the fan controller in my script by changing the fan toggle at the top of the script.

Nokomis449, 2017-10-21:
[screenfan (luma update).zip](https://github.com/cyoung/stratux/files/1403794/screenfan.luma.update.zip)

Here's my script, updated to work with the luma update.  Note that the screen I'm currently using needs the SH1106 driver on line 76 and not the SSD1306 driver, so you'll need to adjust accordingly (the .96" screen uses SSD1306).

Snowflake6, 2017-10-21:
I wonder if the driver needed could be detected somehow at startuo...

Nokomis449, 2017-10-21:
[image]

That'd be nice if it could.  Here's a pic of both my screens running together on the same Stratux box - .96" on left, and a 1.3" on the right.  It's using the SH1106 driver which seems to be running both screens, but the smaller screen is fuzzy and has been for a few months, so I don't know if the SH1106 is its problem or what.  But the larger screen definitely does not work with the SSD1306 driver, so the user will have to somehow select the driver.  Still haven't got the 1.5" screen switched to i2c so I can try it.

Snowflake6, 2017-10-21:
@Nokomis449, you can address both screens on different addresses, i'd be surprised if you couldn't address them with different drivers as well.  You may be having some side effects from trying to run them both on the same address.

commit 0d4d61cf (2017-10-17): Update oled package to new 'luma' package name.

Fixes #672.