cyoung/stratux #673: SSH disabled by default

Snowflake6, 2017-10-16:
1. Stratux version:  1.4r3
2. Stratux config:
not relevant

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
not relevant

4. Description of your issue:

SSH is disabled by default, which means no configuration can be done without a keyboard/monitor available and connected.

Options:

1. Enable SSH by default.  Least secure, but simplest.

2. Enable SSH on first boot, but disabled after boot.  ssh would work until the unit is rebooted, and subsequent boots would be protected.  This would allow detailed configuration on first boot if desired, but better protect the unit once it's set up.  If one wants it enabled permanently one could block the disabling step during that first boot.

Joelbear5, 2017-10-16:
After writing the image to the card, you can add a file with the filename "ssh" to the root of the SD card and the Pi will boot with SSH enabled.

cyoung, 2017-10-16:
@Snowflake6 - SSH should be enabled by default. Just burned the image and SSH'd in.

Could you please confirm the image you wrote to the SD has this MD5:

MD5 (stratux-v1.4r3-4de2420803.img) = a45aa3cdbf97fbd8d75114c9a0ba689e

Snowflake6, 2017-10-16:
Gah!  My sincere apologies.  The last time I built a unit a couple of months ago I had to connect a monitor and keyboard to get SSH activated on it... I meant to post this request back then and forgot.  I made the mistake of assuming this situation hadn't changed.

closed 2017-10-16 by Snowflake6